openthread/wpantund #10: wpantund fails to form a network

GiedriusM, 2016-06-16:
Run `sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/posix/app/ncp/ncp 2" -o SyslogMask "all" -o Config:TUN:InterfaceName vpan0` to start wpantund service.

Run `sudo wpanctl -I vpan0` and try to form a new network:

```shell
wpanctl:vpan0> form ABCD 
Forming WPAN "ABCD" as node type 2
form: error: Did not receive a reply. Possible causes include: the remote applicati
on did not send a reply, the message bus security policy blocked the reply, the rep
ly timeout expired, or the network connection was broken.
Error 11 (0x0B)
wpanctl:vpan0>
````

The error pops up after timeout period. Executing this again finishes immediately with a success status:
```
wpanctl:vpan0> form ABCD
Forming WPAN "ABCD" as node type 2
Successfully formed!
wpanctl:vpan0>
```

darconeous, 2016-06-20:
@GiedriusM: Thanks for filing this issue!

I believe this has been fixed by [OpenThread Pull Request #165](https://github.com/openthread/openthread/pull/165). I've just verified on my local instance. Mind giving it another try with the `master` branches from `wpantund` and OpenThread?

darconeous, 2016-06-21:
Closing for now. @GiedriusM, please reopen if this is still a problem.

closed 2016-06-21 by darconeous