openthread/wpantund #12: Document Windows support

beriberikix, 2016-06-23:
Someone asked if `wpantund` supports Windows. As windows is a popular dev env for embedded tools, it would be useful to document the current windows support and any available future plans.

darconeous, 2016-06-23:
There are currently no plans to support Windows natively, but we wouldn't automatically refuse a pull request that adds support for it.

That being said... [Windows 10 has a new Linux-compatible subsystem](https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux). If it implements support for TUN interfaces, then `wpantund` might just work as-is.

darconeous, 2016-06-23:
You filed the pull request against your own fork. You need to file the pull request against `openthread/wpantund`.

beriberikix, 2016-06-23:
Gah, n00b.

darconeous, 2016-06-30:
Will keep this task open for tracking Windows support.

beriberikix, 2016-09-20:
Once openthread/openthread#629 is merged, can we call this closed? Or is there work that needs to be done on the wpantund side?

nibanks, 2016-09-20:
The changes I have added for Windows support so far are just to get the core libraries building. There is a significant amount of work to get wpantund working on Windows.

darconeous, 2016-09-20:
I honestly don't think there is any good reason to port wpantund to Windows, especially given the work that @nibanks is doing on adding native support to OpenThread.

beriberikix, 2016-09-20:
Makes sense - should we move this to wontfix or leave it open _just in case_?

darconeous, 2016-09-20:
wontfix.

closed 2016-09-20 by darconeous