openthread/wpantund #19: Not able to ping the cli interface on debian when following OpenThread Simulator Tutorial

pszkotak, 2016-07-06:
Hey Team,

I'm running a fresh amd64 debian on a vm.
```
pisz@debian:~$ uname -a
Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2+deb8u3 (2016-07-02) x86_64 GNU/Linux
```

Latest versions of openthread and wpantund are being used, please find commit SHAs below.
openthread used: 8ef67d223be3bac7bfb15948fa254b4fe11c3abf
wpantund used: c0ffa2ab66a73ccb7471c4639f15bb9208f00801

wpantund configure step: 
```
pisz@debian:~/wpantund$ ./configure --sysconfdir=/etc
 ...

= Summary =====================================================================

ConnMan Plugin ................. no
Using libreadline .............. yes
NCP plugins to build ........... spinel dummy
Default NCP Plugin ............. spinel
Static link NCP plugin ......... no
```
As I am not able to find any reference regarding this part I don't know if I should care about missing plugins.

I just want to use only the simulator (there are no NCP connected) so I skipped the following:
`sudo /usr/local/sbin/wpantund -o NCPSocketName /dev/ttyUSB0`

In case it could be somehow related - after I tried to start the wpantund for the first time the following was observed (reboot solved the issue):
```
root@debian:/home/pisz/openthread# /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/apps/ncp/ot-ncp 2" -o Daemon:SyslogMask "all" -o Config:TUN:InterfaceName utun6
wpantund[12750]: Starting wpantund 0.07pisz.00 (Jul  6 2016 09:41:12) . . .
wpantund[12750]: 	SOURCE_VERSION = 0.07.00-3-gc0ffa2a-dirty
wpantund[12750]: 	BUILD_VERSION = 0.07.00-3-gc0ffa2a-dirty
wpantund[12750]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[12750]: Configuration file "/etc/wpantund.conf" read.
wpantund[12750]: set-config-param: "Config:NCP:SocketPath" = "system:examples/apps/ncp/ot-ncp 2"
wpantund[12750]: set-config-param: "Config:TUN:InterfaceName" = "utun6"
wpantund[12750]: set-config-param: "Daemon:SyslogMask" = "all"
wpantund[12750]: set-config-param: "Daemon:SyslogMask" set succeded
wpantund[12750]: Getting DBus connection
wpantund[12750]: Registering DBus connection
wpantund[12750]: Requesting DBus connection name com.nestlabs.WPANTunnelDriver
wpantund[12750]: DBUSIPCServer.cpp:94: Requirement Failed (Connection ":1.42" is not allowed to own the service "com.nestlabs.WPANTunnelDriver" due to security policies in the configuration file)
wpantund[12750]: Runtime error thrown while starting up, "Connection ":1.42" is not allowed to own the service "com.nestlabs.WPANTunnelDriver" due to security policies in the configuration file"
wpantund[12750]: Cleaning up. (gRet = 4)
wpantund[12750]: Stopped.
root@debian:/home/pisz/openthread#
```

Let's proceed to the exact steps that you have to follow to observe the issue:

Terminal A:
```
pisz@debian:~/openthread$ examples/apps/cli/ot-cli 1

> 
> networkname wpantund-test
Done
> extpanid DEBA7AB1E5EAF00D
Done
> channel 14
Done
> masterkey 00112233445566778899AABBCCDDEEFF
Done
> start
Done
> state
[... 140 lines cut ...]
64 bytes from fe80::e0b3:5405:90b0:fd87: icmp_seq=5 ttl=64 time=0.087 ms
^C
--- ff02::1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4000ms
rtt min/avg/max/mdev = 0.087/0.099/0.139/0.020 ms
root@debian:/home/pisz# ping6 ff03::1 -I utun6
ping6: Warning: source address might be selected on device other than utun6.
PING ff03::1(ff03::1) from fe80::a00:27ff:fed5:702 utun6: 56 data bytes
64 bytes from fdde:ba7a:b1e5:0:9704:6a73:208:7028: icmp_seq=1 ttl=64 time=0.874 ms
64 bytes from fdde:ba7a:b1e5:0:9704:6a73:208:7028: icmp_seq=2 ttl=64 time=0.085 ms
64 bytes from fdde:ba7a:b1e5:0:9704:6a73:208:7028: icmp_seq=3 ttl=64 time=0.095 ms
64 bytes from fdde:ba7a:b1e5:0:9704:6a73:208:7028: icmp_seq=4 ttl=64 time=0.087 ms
64 bytes from fdde:ba7a:b1e5:0:9704:6a73:208:7028: icmp_seq=5 ttl=64 time=0.088 ms
^C
--- ff03::1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4001ms
rtt min/avg/max/mdev = 0.085/0.245/0.874/0.314 ms
```

In two last tries only Local and Global addresses assigned to utun6 replied. 
There is no reply from the simulated mesh device.
It happens every time, same symptoms observed.

Please advise, how I can proceed further with this issue?
In case you would like to reproduce the issue and some info are missing, please let me know.

darconeous, 2016-07-07:
Hello @pszkotak!

> As I am not able to find any reference regarding this part I don't know if I should care about missing plugins.

Your configure summary looks fine.

> I just want to use only the simulator (there are no NCP connected) so I skipped the following:
`sudo /usr/local/sbin/wpantund -o NCPSocketName /dev/ttyUSB0`

Where did you get that line? I'm pretty sure that's not in the [tutorial](https://github.com/openthread/wpantund/wiki/OpenThread-Simulator-Tutorial).

> after I tried to start the wpantund for the first time the following was observed (reboot solved the issue):

We've occasionally noticed D-Bus getting in a weird state and not liking `wpantund` until after a reboot. We've only ever seen this on our desktop machines, we've never seen it happen on our embedded devices.

> We see that IP adresses differ from the tutorial

Yeah, the address values appear to be only mildly stable across platforms, but what is weird is `fe80:0:0:0:0:ff:fe00:3000` vs `fe80:0:0:0:0:ff:fe00:0`. @jwhui, any idea where that `3000` might have come from?

> Is `wpantund[1794]: Error fetching property 96 from NCP: 2` expected?

At the moment, this is expected behavior. That is [`PROP_IPv6_LL_ADDR`](https://github.com/darconeous/openthread/blob/308f75ecc4a5a05e3a6fc44142cc96f6e8e120f9/src/ncp/PROTOCOL.md#d51-prop-96-prop_ipv6_ll_addr), which isn't currently implemented on the NCP (2 = [`STATUS_UNIMPLEMENTED`](https://github.com/darconeous/openthread/blob/308f75ecc4a5a05e3a6fc44142cc96f6e8e120f9/src/ncp/PROTOCOL.md#6-status-codes)). It is a harmless red herring.

> Logs observed on Terminal B:
> ```
> wpantund[1794]: State change: "offline" -> "associating"
> wpantund[1794]: "fdde:ba7a:b1e5:0:9704:6a73:208:7028" was added to "utun6"
> wpantund[1794]: Network Name: [...]

That looks kinda odd. 

> There is no reply from the simulated mesh device.
> It happens every time, same symptoms observed.
> Please advise, how I can proceed further with this issue?

First, make sure you don't have any stray copies of `ot-cli` or `ot-ncp` running. If you do, they can interfere with things and cause similar behavior. This caused me a bit of grief myself.

Once you have ruled that out, you can turn on some debugging in `wpantund` by doing the following:

    wpanctl -I utun6 set SyslogMask all

This will make the current instance of wpantund dump out IP traffic and NCP traffic in the logs. Then try pinging again. You may also want to try pinging the link-local address of the `ot-cli`, just to see if that works:

    ping6 fe80:0:0:0:f3d9:2a82:c8d8:fe43%utun6

Could I also get you to paste in the output from typing `route -6` while `wpantund` is running in this state?

jwhui, 2016-07-07:
> Yeah, the address values appear to be only mildly stable across platforms, but what is weird is fe80:0:0:0:0:ff:fe00:3000 vs fe80:0:0:0:0:ff:fe00:0. @jwhui, any idea where that 3000 might have come from?

The 6 most significant bits of the RLOC is the Router ID.  Earlier versions of OpenThread used a deterministic method for allocating Router IDs to assist with debugging.  However, the spec requires that the Router ID be selected at random and OpenThread has been updated to reflect that.

pszkotak, 2016-07-07:
Hello @darconeous, @jwhui,
Thank you for such a quick replies, I did not expect that!

> Where did you get that line? I'm pretty sure that's not in the tutorial.

It's taken from the [wpantund installation guide]
(https://github.com/openthread/wpantund/blob/master/INSTALL.md#2-start-wpantund).

> First, make sure you don't have any stray copies of ot-cli or ot-ncp running. If you do, they can interfere with things and cause similar behavior. This caused me a bit of grief myself.

I'm sure there were none as this was the first thing I have done after the reboot but I will keep it in mind when performing next tries. 

wpantund started with a command: `/usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/apps/ncp/ot-ncp 2" -o SyslogMask "all"`

During all the time we had no more ot processes than these two.
```
root@debian:/home/pisz# ps -A | grep ot-
 1953 pts/0    00:00:00 ot-cli
 1957 pts/4    00:00:00 ot-ncp

```

checked ipconfig and route -6:

```
root@debian:/home/pisz# ifconfig utun6
utun6     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet6 addr: fe80::e0b3:5405:90b0:fd87/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:172 (172.0 B)

root@debian:/home/pisz# route -6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
fe80::/64                      ::                         U    256 0     0 eth0
fe80::/64                      ::                         U    256 0     0 utun6
::/0                           ::                         !n   -1  1     1 lo
::1/128                        ::                         Un   0   1    13 lo
fe80::a00:27ff:fed5:702/128    ::                         Un   0   1     0 lo
fe80::e0b3:5405:90b0:fd87/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 eth0
ff00::/8                       ::                         U    256 0     0 utun6
::/0                           ::                         !n   -1  1     1 lo
root@debian:/home/pisz# 
```

After executing `join 1` in the wpanctl the following was observed:
```
wpantund[2186]: [NCP->] CMD_PROP_VALUE_IS(PROP_PHY_CHAN)
wpantund[2186]: SELECT: fd_count=8 cms_timeout=60000
wpantund[2186]: SELECT:     read_fd_set: 4, 5, 6, 7
wpantund[2186]: SELECT:     write_fd_set: 
wpantund[2186]: [->NCP] Dropping matched packet.
wpantund[2186]: SELECT: fd_count=8 cms_timeout=59658
wpantund[2186]: SELECT:     read_fd_set: 4, 5, 6, 7
wpantund[2186]: SELECT:     write_fd_set: 
wpantund[2186]: [->NCP] Dropping matched packet.
wpantund[2186]: SELECT: fd_count=8 cms_timeout=59626
wpantund[2186]: SELECT:     read_fd_set: 4, 5, 6, 7
wpantund[2186]: SELECT:     write_fd_set: 
wpantund[2186]: [->NCP] Dropping matched packet.
wpantund[2186]: SELECT: fd_count=8 cms_timeout=56046
wpantund[2186]: SELECT:     read_fd_set: 4, 5, 6, 7
wpantund[2186]: SELECT:     write_fd_set: 
wpantund[2186]: [->NCP] Dropping matched packet.
wpantund[2186]: SELECT: fd_count=8 cms_timeout=52038
[... 297 lines cut ...]
]
```

Now instead entering:
```
> networkname wpantund-test
> extpanid DEBA7AB1E5EAF00D
> channel 14
> masterkey 00112233445566778899AABBCCDDEEFF
```

I do:

```
> networkname wpantund-test
> extpanid DEBA7AB1E5EAF00D
> channel 11
> panid 1234
> masterkey 00112233445566778899AABBCCDDEEFF
```

Changed panid as well, I am not sure if 0xFFFF is an allowed value.
After these changes I am able to ping the simulated device.

Thank you for your involvement, I'm waiting for further comments.

jwhui, 2019-04-23:
Closing stale issue.

closed 2019-04-23 by jwhui