openthread/wpantund #24: Creating Thread network in simulator

Praveen-Kumar-K, 2016-07-20:
Hi all,
I want to simulate Thread network on my PC, i have followed wpantund tutorial mentioned in repo.
I have used 4 clients (1 Leader and 3 routers assigned), one Wpantund, one utun6.
clients haveing parameters as shown:
cl1: name----wpantund-test; ExPANID-----deba7ab1e5eaf00a; PAN-----FFFF; Chaneel---11;
cl2: name----wpantund-test; ExPANID-----deba7ab1e5eaf00b; PAN-----FFFF; Chaneel---11; -leader
cl3: name----wpantund-test; ExPANID-----deba7ab1e5eaf00c; PAN-----FFFF; Chaneel---11;
cl4: name----wpantund-test; ExPANID-----deba7ab1e5eaf00d; PAN-----FFFF; Chaneel---11;
For Wpantund terminal i have done command.

sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/apps/ncp/ot-ncp 2" -o SyslogMask " -info" -o Config:TUN:InterfaceName utun6
[sudo] password for pkamma:
wpantund[320]: Starting wpantund 0.07.00 (Jul 13 2016 17:54:13) . . .
wpantund[320]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[320]: Configuration file "/etc/wpantund.conf" read.
wpantund[320]: CONFIGURATION PROPERTY "SyslogMask" IS DEPRECATED. Please use "Daemon:SyslogMask" instead.
wpantund[320]: set-config-param: "Config:NCP:SocketPath" = "system:examples/apps/ncp/ot-ncp 2"
wpantund[320]: set-config-param: "Config:TUN:InterfaceName" = "utun6"
wpantund[320]: set-config-param: "Daemon:SyslogMask" = " -info"
wpantund[320]: Ready. Using DBUS bus ":1.21"
wpantund[322]: About to exec "examples/apps/ncp/ot-ncp 2"
wpantund[320]: Running as root without dropping privileges!
wpantund[320]: State change: "uninitialized" -> "offline"
wpantund[320]: NCP is running "OPENTHREAD/0.01.00; Jul 14 2016 10:59:06"
wpantund[320]: Driver is running "0.07.00 (; Jul 13 2016 17:54:13)"
wpantund[320]: "fe80::e0b3:5405:90b0:fd87" was added to "utun6"
wpantund[320]: Network Name: OpenThread
wpantund[320]: **Error fetching property 96 from NCP: 2**
wpantund[320]: Finished initializing NCP

in Utun6 terminal: When applying scan i am not able to see clients.

To proceed further, to form a network what should i do, can any one help me in this regard.

Thanks,
Pravee Kumar K

darconeous, 2016-08-02:
This is the same as [OpenThread issue #271](https://github.com/openthread/openthread/issues/271), right?

Praveen-Kumar-K, 2016-08-03:
Yes @darconeous

darconeous, 2016-09-30:
Closing, since [OpenThread issue #271](https://github.com/openthread/openthread/issues/271) has been closed.

closed 2016-09-30 by darconeous