openthread/wpantund #63: DBUSIPCServer.cpp

sudharshanredd, 2016-09-23:
when iam configuring wpantund-master i was geeting error at DBUSIPCServer.cpp:93:Requiremet Failed(Connection ":1.74" is not allowed to own the service "com.nestlabs.WPANTunnelDriver" due to security policies in the configuration file).

Steps following to configure:
./bootstrap.sh
./configure --sysconfdir=/etc
make

when iam going to compile in wpantund it is throughing error like DBUSIPCServer.cpp.
can you please look into the issue and give some information to debug.

Thanks,
Sudharshan.
[image]

darconeous, 2016-09-27:
The default `sysconfdir` should be `/etc`, even if you don't specify it directly. So if you install it where it should be, you should be all set.

That being said, sometimes D-Bus seems to get into a weird state where it starts behaving like this. A reboot seems to fix it. Worth a try.

You didn't mention what your specific build error was on your laptop. Mind pasting it in?

darconeous, 2016-12-01:
Closing. Please re-open if you can provide more details.

closed 2016-12-01 by darconeous