openthread/wpantund #65: Building wpantund may fail due to wpan-tunnel-plugin.la naming convention

pszkotak, 2016-09-26:
Automake complains about the library name:
`'wpan-tunnel-plugin.la' is not a standard library name; did you mean 'libwpan-tunnel-plugin.la'`

Two solutions can be proposed:
1) Changing the name 'wpan-tunnel-plugin.la' to libwpan-tunnel-plugin.la.
I'm using own patch to do it at this moment, but this solution is not clean enough.
2) Adding '-module' parameter to the wpan_tunnel_plugin_la_LDFLAGS in Makefile.am.
It seems that nl.m4 assigns this value in line 272:
`then CONNMAN_LIBS="-module -avoid-version -export-symbols-regex connman_plugin_desc"`
however it was not executed in my scenario when trying to cross compile wpantund. Which conditions have to be met to have '-module' assigned?

darconeous, 2016-09-27:
Building won't fail. It's just a warning, and a fairly annoying one at that. In this case we can easily shut it up by putting `-module` in the Makefile.

That plugin only attempts to get made if you have ConnMan on your system.

pszkotak, 2016-09-28:
@darconeous, thank you for an answer.
Well, the build fails in my case. The Makefile.in doesn't get created, I didn't dig deeper.
Putting `-module` solves the issue.

I will submit a PR doing that.

PR #71 (2016-09-28): Add missing -module LDFLAG
This PR fixes #65
closed 2016-09-28 by darconeous