openthread/wpantund #81: Using wpantund as Border Router

IOTUser, 2016-10-14:
I am trying to connect my devices running openthread to my Network to be able to access them using IPv6.

I have one board running OpenThread CLI (UART connected to PC) and one board running OpenThread NCP (UART connected to Raspberry Pi)

I am able to form a Network and can ping using Link-Local IPv6-Adresses from the OT-CLI and from the Linux Shell.

Issue 1: While OT-CLI Shows the mesh-local addresses using "ipaddr", "ip addr" on Linux returns only the Link Local Address. Adding the address via "ip addr add fdde:ad00:beef:0:0:ff:fe00:9400/64 dev wpan0" (as displayed in      "IPv6:MeshLocalAddress" => "fdde:ad00:beef:0:6610:983c:bd88:c3a0"   "IPv6:MeshLocalPrefix" => "fdde:ad00:beef::/64") results in working pings to this mesh local Address. Shouldn't this address be added to the Interface by wpantun automatically? This behavior is identical on simulator.

Question 2: I would like to reach the OpenThread-Network via LAN. Since I don't have external IPv6 I configured site-local IPs on the raspberry and a connected computer and added a route to the mesh-local prefix via the raspberry and enabled ipv6 forwarding there  In the wpantund-console I can see the Pings being transmitted (with site-local as sender and mesh-local as receiver). But I do not get a answer. Do I have to use "config-gateway" or "add-route" to announce this route to the thread-network?

IOTUser, 2016-11-04:
I got it working with few devices:
I configured all devices with OT-CLI, setting Panid ifconfig up and thread start direct in the firmware. 
When using 2-3 Devices it took some time for the border router to get promoted to "router"-role from its "end-device"-role (even though I was joining using "join -Tr 1")
Now Pings and UDP-Packets got forwarded in both directions.

When adding more boards the border-router-NCP reverts to its "end-device"-role and stops routing packets. Is there any way to force the border router to use the "router"-role?

IOTUser, 2016-11-04:
In some cases the Border Router stays in his "router"-role, but ot-cli-devices became "end-devices". 
They stop receiving packets addressed to the external prefix, while using the mesh-local address still works.

rmadhuraj, 2017-03-06:
Hi ..
          Can you tell me how to make NCP as border router. I tried, but the NCP is not able to join more than one Network at a time. When i try to join NCP with 2nd network , it is throwing an error **join failed with error 10. Already**.
Is there any procedure to create the network with boarder router and connect to other networks ??
My Requirement is like collect the data from end-device and send it to the cloud.

Thanks
Madhu Raj.

beriberikix, 2017-03-06:
Please use [wpantund-users](https://groups.google.com/forum/#!forum/wpantund-users) for discussions around syntax and "how-to." Github issues are meant for issues and development of `wpantund`.

closed 2017-03-06 by beriberikix