openthread/wpantund #83: Mesh Local Address does not get assigned to wpan-Interface

IOTUser, 2016-10-19:
Mesh Local IPv6 Adresses not added to wpan0 interface.
Steps to reproduce:

One Board running openthread-cli.
panid 0x1234
ifconfig up
thread start

Second board connected via UART to Raspberry:
In the wpanctl-console:
scan

   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
 1 |       NO | "OpenThread"       | 0x0123 | 11 | DEAD00BEEF00CAFE | 22C27CA5BBA8ECF7 |    0

join 1

Joining "OpenThread" DEAD00BEEF00CAFE as node type "end-device"
Successfully Joined!

To get the IP of this system: 
exit wpanctl, then:
ip addr
[...]
3: wpan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1280 qdisc fq_codel state UNKNOWN group default qlen 500
    link/none
    inet6 fe80::88ce:5314:41c0:f51d/64 scope link
       valid_lft forever preferred_lft forever

After adding the address via "ip addr add fdde:ad00:beef:0:0:ff:fe00:9400/64 dev wpan0" (as displayed in wpanctl status under "IPv6:MeshLocalAddress" => "fdde:ad00:beef:0:6610:983c:bd88:c3a0" "IPv6:MeshLocalPrefix" => "fdde:ad00:beef::/64")

Pings to this mesh local address. works from the OT-CLI board, Pings to the mesh local address of the ot-cli-board now work too.

Shouldn't this address be added to the Interface by wpantun automatically? This behavior is identical on simulator-devices instead of real hardware.

darconeous, 2016-10-28:
Yes it should, and it does seem to be being added on my systems. Will investigate.

IOTUser, 2016-11-02:
OS is Arch Linux. After using "config-gateway -d fec01000" a address with prefix fec0:1000 is added to this interface successfully.

darconeous, 2016-12-01:
This may have been addressed in recent changes. Can you still reproduce this?

shahab5, 2016-12-08:
I also faced this issue with Fedora Core 22, earlier versions of wpantund used to work fine.
The test is carried on simulator `ot-cli-ftd` and `ot-ncp`.

Thanks,
Shahab

jwhui, 2016-12-09:
#112 may address this issue.

darconeous, 2016-12-14:
@IOTUser, see if #112 (just merged) helps.

closed 2019-04-23 by jwhui