openthread/wpantund #89: Wpantund fails to build

pszkotak, 2016-10-28:
Hi Team,

wpanctl.c fails to compile with the following message when HAVE_LIBREADLINE is defined.

```
wpanctl.c: In function 'main':
wpanctl.c:485:10: error: assignment of read-only variable 'stdin'
    stdin = fopen(optarg, "r");
          ^
make[7]: *** [wpanctl-wpanctl.o] Error 1 
```

darconeous, 2016-10-28:
What platform are you building on?

pszkotak, 2016-10-31:
I'm cross-compiling on x86 with openwrt build system.
Target is the cortex A53, gcc 5.3.0 used, in case that matters.

darconeous, 2016-11-30:
Ah. Well, it looks like they've made `stdin` constant on your platform. C89 does allow implementations to make `stdin` not be an lvalue, so this is indeed a bug.

[The fix for this is to use `freopen()`](http://stackoverflow.com/questions/584868/rerouting-stdin-and-stdout-from-c) instead of `fopen()`, which seems pretty straightforward.

darconeous, 2016-12-01:
@pszkotak, Please give the proposed fix in #110 a try.

PR #110 (2016-12-01): Do not use stdin or stdout as an l-value
In a handful of places, `stdin` and `stdout` were being used as l-values. While `stdin` and `stdout` can be used as an l-value in many standard library implementations, it is explicitly stated in C89 that it is acceptable for implementations to not make these symbols l-values. Issue #89 is an example of this.

This change rewrites all places where we are using these variables as l-values so that `wpantund` can be used on those platforms.

This commit fixes and closes #89.
commit b352796c (2016-12-01): Do not use stdin or stdout as an l-value

In a handful of places, `stdin` and `stdout` were being used as
l-values. While `stdin` and `stdout` can be used as an l-value
in many standard library implementations, it is explicitly stated
in C89 that it is acceptable for implementations to not make
these symbols l-values. Issue #89 is an example of this.

This change rewrites all places where we are using these variables
as l-values so that `wpantund` can be used on those platforms.

This commit fixes and closes #89.
closed 2016-12-01 by darconeous