openthread/wpantund #136: Update paths in install.md and OpenThread Simulator Tutorial

beriberikix, 2017-02-28:
Found a few things to fix in the docs but wanted to raise them before I tried to fix:

# install.md
- `d-bus` [renamed](https://github.com/Homebrew/homebrew-core/pull/1849) to `dbus`
- Homebrew on Sierra (at least on my machine; I could have mucked with paths) places Cellars elsewhere and `cp` fails. For me it was `cat /usr/local/Cellar/dbus/*/org.freedesktop.dbus-session.plist`
  - therefore should update `cp` [command](https://github.com/openthread/wpantund/blob/master/INSTALL.md#installing-wpantund-on-os-x) to start the daemon and update to `brew install dbus`

# OpenThread Simulator Tutorial
- [Preparation](https://github.com/openthread/wpantund/wiki/OpenThread-Simulator-Tutorial#preparation) section could move to the convenience makefile for [Posix](https://github.com/openthread/openthread/blob/master/examples/Makefile-posix)
- ot-cli is now at `output/x86_64-apple-darwin/bin/ot-cli-ftd` as [referenced](https://github.com/openthread/wpantund/wiki/OpenThread-Simulator-Tutorial#starting-the-cli-app) in Starting the `cli` app
- ot-ncp is now at `output/x86_64-apple-darwin/bin/ot-ncp-ftd` as [referenced](https://github.com/openthread/wpantund/wiki/OpenThread-Simulator-Tutorial#starting-wpantund) in `Starting wpantund`

I still haven't been able to join using `wpanctl` so there might be something I did incorrectly above.

bukepo, 2017-03-07:
Did you get the fix of this issue? @beriberikix

bukepo, 2017-03-07:
I got this working on macOS Sierra. If you are installing with Homebrew, the failure is probably caused by the mismatching NCP version. Try installing the latest version with
```
brew uninstall wpantund
brew install ./etc/wpantund.rb --HEAD
```

closed 2019-04-23 by jwhui