openthread/wpantund #137: How to add nodes to NCP network ?

rmadhuraj, 2017-03-02:
Hi all,
          I followed the **Open thread simulator tutorial**.
          I am able to join the NCP as end-device to a CLI-network. 
          I tried forming a network with NCP-wpantund (untun6>form "abc") , the network is formed . But when i tried to join a CLI node to the NCP (assigned the PAN-id of NCP network to cli node ), the new cli node is becoming a leader. and i see no interaction between CLI and NCP. I am confused . 
            Is it right procedure to do so.? How it is possiible ? Can you clarify how they will communicate.

Thanks and Regards.
Madhu Raj

jwhui, 2017-03-02:
As part of forming a new network, wpantund will configure a new PAN ID,
Extended PAN ID, and Master Key.  When trying to attach a CLI node to the
NCP, you will need to make sure all of the network parameters match.  On
the wpantund side, you can use the wpanctl `status` command to view the
network parameters.

Hope that helps.

rmadhuraj, 2017-03-03:
Yes it worked. While forming the cli node ,i set the parameters ( PANID,Masterkey and channel) matching with NCP network.( EXPANID is unchangeable,still cli node joined the network).

Thanks @jwhui

rmadhuraj, 2017-03-03:
Further investigating ,
             Initially i created cli netwok and made NCP to join the cli netwok, Joined successfully.
 After sometime i made NCP to Leave the current Network , and tried to form a new network with NCP.
 it is failed , i tried many times , it fails every time to form a network.
 
What may be went wrong ?. do i need to set any Property. I even tried Resetting the NCP , its not able to form the network.

**wpanctl:utun6> join 3
Joining "OpenThread" DEAD00BEEF00CAFE as node type "end-device"
Successfully Joined!**

**wpanctl:utun6> leave
Leaving current WPAN. . .
wpanctl:utun6> form "aaaaa"
Forming WPAN "aaaaa" as node type "router"
form: error: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Error 11 (0x0B)**

Thanks 
Madhu Raj.

rmadhuraj, 2017-03-31:
This issue i didn't observe when i ran the the applications on Boards.
the Reset command on Posix is not working , But it is able to reset on the Board.

Thanks and Regards
R Madhu

closed 2017-03-31 by rmadhuraj