openthread/wpantund #138: Form command overwrites network configuration

GiedriusM, 2017-03-02:
After executing the following:
```
setprop Network:Key 00112233445566778899aabbccddeeff
setprop Network:PANID 0x8888
setprop Network:XPANID cafe00d00d00cafe
form -c 11 TEST
```
PANID and XPANID are not as specified, but rather randomly generated. SpinelNCPTaskForm checks the passed options for these values and if they are not passed - it generates new random ones. Tracing back to DBusIPCAPI_v0::interface_form_handler I can see that these values are not set there, which means that PANID and XPANID will always be overwritten.
However with network key it's slightly different - they are randomly generated only if they were not set before. IMO it should be like this with PANID and XPANID too.

Is this a bug or is there a reason for inconsistent behavior?

darconeous, 2017-03-14:
Currently, you need to specify those parameters at the form command line. Yes, it is inconsistent.  We will likely move the master key to that soon as well.

We will be moving away from using properties in this way at some point (maybe for v0.08.x?)

I'll be making some documentation updates soon for how all of this should work.

darconeous, 2017-04-21:
Converted to a task and updated the issue name.

PR #158 (2017-05-08): Form network based on existing PANID and XPANID
Setting PSKc relies on forming network with given XPANID. This PR fixes #138 .
closed 2017-05-08 by darconeous