openthread/wpantund #156: if /bin/sh = dash, configure fails

DuaneEllis-TI, 2017-04-24:
Default shell on debian platforms is /bin/dash

And when using dash, wpantund configure script fails with this error message.

```
checking for getdtablesize... yes
checking for getloadavg... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for DBUS... yes
./configure: line 17700: syntax error near unexpected token `-std=c++11,'
./configure: line 17700: `                                      AX_CHECK_COMPILE_FLAG(-std=c++11,'
```
Options: Require BASH or Fix this.

embeddedRSA, 2019-09-20:
This is also the case for Ubuntu 18.05 bionic... I suspect that libreadline and libreadline-dev used to be too separate packages.. either way my wpantund says that libreadline is active by just getting libreadline-dev.

silver2row, 2022-02-14:
Hello...

I am building on a similar armhf board: BBGG. Also from beagleboard.org but here is my issue:

```

Making all in ipc-dbus
make[3]: Entering directory '/home/debian/wpantund/src/ipc-dbus'
  CXXLD    libwpantund-dbus.la
libtool:   error: 'libwpantund_dbus_la-DBUSIPCServer.lo' is not a valid libtool object
make[3]: *** [Makefile:565: libwpantund-dbus.la] Error 1
make[3]: Leaving directory '/home/debian/wpantund/src/ipc-dbus'
make[2]: *** [Makefile:453: all-recursive] Error 1
make[2]: Leaving directory '/home/debian/wpantund/src'
make[1]: *** [Makefile:392: all] Error 2
make[1]: Leaving directory '/home/debian/wpantund/src'
make: *** [Makefile:534: all-recursive] Error 1

```

Seth

P.S. If you are still interested in making wpanctl work w/ the BeagleBone Black, please do reply.