openthread/wpantund #162: Add cross-compilation instructions

beriberikix, 2017-05-09:
To start, it would be great to have instructions on how to x-compile wpantund on MacOS/Linux targeting a less-beefy SBC like the Raspberry Pi 3. As a user I've used http://crosstool-ng.github.io/ but I'm sure there are other options out there, especially using automake.

darconeous, 2017-05-09:
Yes, one of the downsides to using boost is the crazy resources used while compiling. :(

DuaneEllis-TI, 2017-05-24:
the other issue with this type of cross is you need more then the compiler... you need all of the secondary packages and their related header files in your cross build environment.

It's non-trivial to set this up, and I think it is well beyond the scope of 'wpantund' to document this.

d-bahr, 2017-06-02:
I have successfully cross-compiled wpantund and wpanctl for a BeagleBone Black via arm-linux-gnueabihf-gcc. Here are the steps. If there's a better place to put this write-up (like on a Wiki page) I would be happy to drop it there.

First, in a clean repo:
`sudo bash ./bootstrap.sh`

Don't forget to sudo-alternatives the arm-linux-gnueabihf gcc installation (it doesn't seem to do it by default on my installation):
https://askubuntu.com/questions/26498/choose-gcc-and-g-version

```
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-gcc arm-linux-gnueabihf-gcc /usr/bin/arm-linux-gnueabihf-gcc-4.8 10
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-g++ arm-linux-gnueabihf-g++ /usr/bin/arm-linux-gnueabihf-g++-4.8 10
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-cpp arm-linux-gnueabihf-cpp /usr/bin/arm-linux-gnueabihf-cpp-4.8 10
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-gcc-ar arm-linux-gnueabihf-gcc-ar /usr/bin/arm-linux-gnueabihf-gcc-ar-4.8 10
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-gcc-nm arm-linux-gnueabihf-gcc-nm /usr/bin/arm-linux-gnueabihf-gcc-nm-4.8 10
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-gcc-ranlib arm-linux-gnueabihf-gcc-ranlib /usr/bin/arm-linux-gnueabihf-gcc-ranlib-4.8 10
sudo update-alternatives --install /usr/bin/arm-linux-gnueabihf-gcov arm-linux-gnueabihf-gcov /usr/bin/arm-linux-gnueabihf-gcov-4.8 10
```

Need to download dbus arm library for armhf builds:
```
sudo dpkg --add-architecture armhf
sudo apt-get update
```

Add these lines to /etc/apt/sources.list:
```
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty-updates main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty-backports main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty-security main
```

Then:
`sudo apt-get update`

Go to a temporary directory (important!) and run:
```
sudo apt-get download libdbus-1-dev:armhf
sudo apt-get download libreadline6:armhf
sudo apt-get download libreadline6-dev:armhf
sudo apt-get download libreadline-dev:armhf
sudo dpkg -i libdbus-1-dev_1.4.18-1ubuntu1_armhf.deb
sudo dpkg -i libreadline6_6.3-4ubuntu2_armhf.deb
sudo dpkg -i libreadline6-dev_6.3-4ubuntu2_armhf.deb
sudo dpkg -i libreadline-dev_6.3-4ubuntu2_armhf.deb
```

The above lines may be different depending on the name of the packages that were downloaded. Also, it may give some errors, but you can safely ignore those. You can delete the temporary directory when you are done.

Finally:
`export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/usr/lib/arm-linux-gnueabihf/pkgconfig`

And the moment you've been waiting for (it can be built now):
```
./configure --sysconfdir=/etc --host=arm-linux-gnueabihf ac_cv_search_readline=/usr/lib/arm-linux-gnueabihf/libreadline.so CC=/usr/bin/arm-linux-gnueabihf-gcc CXX=/usr/bin/arm-linux-gnueabihf-g++ CCLD=/usr/bin/arm-linux-gnueabihf-ld CXXLD=/usr/bin/arm-linux-gnueabihf-ld CFLAGS="$(pkg-config dbus-1 --cflags)"
make -j12
make install
```

Note: You may need to use this configure step instead (depends on linking/version of D-Bus):
`./configure --sysconfdir=/etc --host=arm-linux-gnueabihf ac_cv_search_readline=/usr/lib/arm-linux-gnueabihf/libreadline.so CC=/usr/bin/arm-linux-gnueabihf-gcc CXX=/usr/bin/arm-linux-gnueabihf-g++ CCLD=/usr/bin/arm-linux-gnueabihf-ld CXXLD=/usr/bin/arm-linux-gnueabihf-ld CFLAGS="$(pkg-config dbus-1 --cflags) -pthread" LDFLAGS="-pthread"`

The -j12 option sometimes fails. You may need to drop that number (-j4 works nicely) or not use the "-j" option at all.

Also, you can remove the dpkg arm architecture after you're done installing dbus, if the sudo apt-get error messages annoy you.
Follow this link to find and remove libdbus-1-dev (armhf architecture!):
https://askubuntu.com/questions/85748/how-can-i-remove-the-record-of-an-installed-package-without-deleting-any-files
Open /var/lib/dpkg/status. Search for libdbus-1-dev and remove that section (again, armhf architecture only!). Recommend saving a copy of the file as a backup in case something goes wrong.

[Following steps are optional, only the above steps are necessary to remove the apt-get error message]
Then:
`sudo dpkg --remove-architecture armhf`

And remove the following lines from /etc/apt/sources.list:
```
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty-updates main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty-backports main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports trusty-security main
```

Lastly, some notes on installing the binaries:

wpantund installs here:
`/usr/local/sbin/wpantund`

With some plugins here:
`/usr/local/libexec/wpantund/…`
MAKE SURE YOU CHMOD +X THESE THINGS!

With configuration file here:
```
/etc/wpantund.conf
/etc/dbus-1/system.d/wpantund.conf
```

There are also some header files here:
`/usr/local/include/wpantund/`

wpanctl installs here:
`/usr/local/bin/wpanctl`

d-bahr, 2017-06-13:
The above steps are for Ubuntu 14.04. I just tried those steps for 16.04, and it failed miserably, but I get it working via a much simpler method, using a simple multiarch setup:

Steps to set up multiarch:
`sudo dpkg --add-architecture armhf`

Add these lines to /etc/apt/sources.list:
```
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports xenial main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports xenial-updates main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports xenial-backports main
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports xenial-security main
```

Then:
`sudo apt-get update`

Finally, install the libraries (armhf versions) need to build wpantund:
```
sudo apt-get install libdbus-1-dev:armhf libdbus-1-3-dev:armhf
sudo apt-get install libreadline-dev:armhf libreadline6-dev:armhf libreadline6:armhf
sudo apt-get install libsystemd0:armhf libsystemd-dev:armhf
```

Configure dbus pkg-config:
`export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/usr/lib/arm-linux-gnueabihf/pkgconfig`

Now you can build away:
```
sudo bash ./bootstrap.sh
./configure --sysconfdir=/etc --host=arm-linux-gnueabihf ac_cv_search_readline=/usr/lib/arm-linux-gnueabihf/libreadline.so CC=/usr/bin/arm-linux-gnueabihf-gcc CXX=/usr/bin/arm-linux-gnueabihf-g++ CCLD=/usr/bin/arm-linux-gnueabihf-ld CXXLD=/usr/bin/arm-linux-gnueabihf-ld CFLAGS="$(pkg-config dbus-1 --cflags) -pthread" LDFLAGS="-pthread"
make -j4
```

darconeous, 2017-06-14:
Nice! I wonder if we could wrap up all of those steps into a docker container so that it would be easier to do.

fabiannydegger, 2018-07-11:
Hello
I've tried to cross-compile wpantund for the Omega2+
But I just don't know how. can you help me?

I installed the build system for openWRT https://openwrt.org/docs/guide-developer/build-system/start
On Ubuntu lts 16.04
built it with:
```
For Target System, MediaTek Ralink MIPS
For Subtarget, MT7688 based boards
For Target Profile, Onion Omega2+
```

And followed https://openwrt.org/docs/guide-developer/crosscompile

**My problem is:** 
where are my files and how do I get them on the omega?

d-bahr, 2018-07-13:
I know very little about the Omega2+, but my guess is that running `xCompile.sh` is overwriting some of the include paths and other compilation configurations created during the `./configure` step for wpantund. I would recommend running `xCompile.sh` with `--debug` to see what it wants to set for LDFLAGS, CFLAGS, CXX, etc. and then manually enter those values into the `./configure` command, then run `make` as you would normally when building wpantund. The only remaining tricky thing would be to figure out what to specify for `--host` and I'm not sure what that might be.

osaether, 2020-09-24:
I am trying to cross compile for Openembedded (Victron Venus GX) and I have gotten quite far but now I am stuck with this error:
`
wpan-connman-plugin.c: In function ‘scan_finished_callback’:
wpan-connman-plugin.c:2194:38: error: ‘CONNMAN_SERVICE_TYPE_LOWPAN’ undeclared (first use in this function); did you mean ‘CONNMAN_SERVICE_TYPE_VPN’?
  connman_device_set_scanning(device, CONNMAN_SERVICE_TYPE_LOWPAN, FALSE);
`

I have searched several connman repositores but have not found any that defines the CONNMAN_SERVICE_TYPE_LOWPAN. All have this struct in the services.h file:

`enum connman_service_type {
	CONNMAN_SERVICE_TYPE_UNKNOWN   = 0,
	CONNMAN_SERVICE_TYPE_SYSTEM    = 1,
	CONNMAN_SERVICE_TYPE_ETHERNET  = 2,
	CONNMAN_SERVICE_TYPE_WIFI      = 3,
	CONNMAN_SERVICE_TYPE_BLUETOOTH = 4,
	CONNMAN_SERVICE_TYPE_CELLULAR  = 5,
	CONNMAN_SERVICE_TYPE_GPS       = 6,
	CONNMAN_SERVICE_TYPE_VPN       = 7,
	CONNMAN_SERVICE_TYPE_GADGET    = 8,
	CONNMAN_SERVICE_TYPE_P2P       = 9,
};
#define	MAX_CONNMAN_SERVICE_TYPES        10`