openthread/wpantund #172: CLI join network created by WPAN?

DuaneEllis-TI, 2017-05-25:
If I use wpanctl to "form" a network...
I can't seem to join the network from an CLI.... device.

In the reverse case, I form a network using a CLI device...
Wpantund/wpanctl *can* join that network.

Something seems wrong, or perhaps there is a setting that is not documented?

jaredwolff, 2017-06-09:
+1 on this. It's hard to know what magic incantation of CLI commands are necessary to connect when the NCP is acting as a router. On the other hand, I was able to get the NCP to connect to the CLI no problem. Have you tried that Duane?

DuaneEllis-TI, 2017-06-09:
Actually, I did.

The magic can be distilled from the wpantund script installed here: `/usr/share/wpantund/wpanretain.sh`
And found in the source here: `src/scripts/wpantund.sh`

Basically, once wpantund starts - this script can be used to query wpantund and write the configuration data to a file (ie: save command) it can also later be used to recall that configuration. There might be some more items that are  not saved/restored but I'm only using mostly defaults today.

The data in those files is what you need to join the network.

Second, wpanctl has a few commands, specifically: `scan` and `join` command, these commands can
be used to scan for existing networks and then join one of the discovered networks. However there is a catch, the join command expects an INDEX number into the scanned table.

Many examples of wpanctl show commands in the standalone command line mode, not interactive mode. For example:
```
$  wpantund -I  wpan0  scan
(which produces output, the result of the scan)
```
The problem is this: When the above scan command is done, the program exits. The next invocation does not have the same table, thus the JOIN command does not work correctly in the second invocation.

These solution is to use wpanctl in interactive mode like this:
```
$ sudo wpanctl -I wpan0
(this launches wpanctl in the interactive mode, you can now enter the scan and join commands)
wpantund-wpan0> scan
(the result of the scan)
wpantund-wpan0> join X
(where X is the index of the table row you wish to join)
```
You MAY need other things, for example the network key and things like that.

closed 2019-04-23 by jwhui