openthread/wpantund #188: "Form" network: Error 7 with NCP UART - Ubuntu 16.04

biboc, 2017-06-13:
Hi, 
Here is my setup:
- NCP connected by UART
- wpantund on Ubuntu 16.04

Communication and initialisation between NCP and wpantund is ok:
```
openthread$ sudo /usr/local/sbin/wpantund -s /dev/pts/11 -I wpan0
wpantund[2453]: Starting wpantund 0.08.00d (May 11 2017 13:39:58) . . .
wpantund[2453]: 	SOURCE_VERSION = 0.07.01-39-gf23affa
wpantund[2453]: 	BUILD_VERSION = 0.07.01-39-gf23affa
wpantund[2453]: Reading configuration from "/usr/local/etc/wpantund.conf" . . .
wpantund[2453]: Configuration file "/usr/local/etc/wpantund.conf" read.
wpantund[2453]: set-config-param: "Config:NCP:SocketPath" = "/dev/pts/11"
wpantund[2453]: set-config-param: "Config:TUN:InterfaceName" = "wpan0"
wpantund[2453]: Ready. Using DBUS bus ":1.78"
wpantund[2453]: Stat (autolog): Tx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[2453]: Stat (autolog): Tx history is empty
wpantund[2453]: Stat (autolog): 
wpantund[2453]: Stat (autolog): Rx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[2453]: Stat (autolog): Rx history is empty
wpantund[2453]: Stat (autolog): 
wpantund[2453]: Stat (autolog): NCP state history is empty.
wpantund[2453]: Stat (autolog): 
wpantund[2453]: Stat (autolog): 
wpantund[2453]: Opening tun interface socket with name "wpan0"
wpantund[2453]: Running as root without dropping privileges!
wpantund[2453]: Initializing NCP
wpantund[2453]: NCP is now BUSY.
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_PROTOCOL_VERSION) tid:1
wpantund[2453]: 	↳ 810201
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_PROTOCOL_VERSION) tid:1
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_NET_STACK_UP) tid:2
wpantund[2453]: 	↳ 820242
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_STACK_UP) tid:2
wpantund[2453]: State change: "uninitialized" -> "offline"
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_NCP_VERSION) tid:3
wpantund[2453]: 	↳ 830202
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NCP_VERSION) tid:3
wpantund[2453]: NCP is running "OPENTHREAD/gfbfd76a-dirty; none; Jun  9 2017 11:17:49"
wpantund[2453]: Driver is running "0.08.00d (0.07.01-39-gf23affa; May 11 2017 13:39:58)"
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_INTERFACE_TYPE) tid:4
wpantund[2453]: 	↳ 840203
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_INTERFACE_TYPE) tid:4
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_VENDOR_ID) tid:5
wpantund[2453]: 	↳ 850204
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_VENDOR_ID) tid:5
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_CAPS) tid:6
wpantund[2453]: 	↳ 860205
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_CAPS) tid:6
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_HWADDR) tid:7
wpantund[2453]: 	↳ 870208
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_HWADDR) tid:7
wpantund[2453]: NCP Status: MACHardwareAddr:   00:00:00:FF:FE:00:FF:FE
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_PHY_CHAN) tid:8
wpantund[2453]: 	↳ 880221
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_PHY_CHAN) tid:8
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_PHY_CHAN_SUPPORTED) tid:9
wpantund[2453]: 	↳ 890222
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_PHY_CHAN_SUPPORTED) tid:9
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_MAC_15_4_PANID) tid:10
wpantund[2453]: 	↳ 8A0236
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_MAC_15_4_PANID) tid:10
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_MAC_15_4_LADDR) tid:11
wpantund[2453]: 	↳ 8B0234
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_MAC_15_4_LADDR) tid:11
wpantund[2453]: NCP Status: MACAddr:           BA:AD:4C:41:2A:C5:92:90
wpantund[2453]: [->NCP] CMD_PROP_VALUE_GET(PROP_NET_MASTER_KEY) tid:12
wpantund[2453]: 	↳ 8C0246
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_MASTER_KEY) tid:12
[... 100 lines cut ...]
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_XPANID) tid:14
wpantund[2453]: [->NCP] CMD_PROP_VALUE_SET(PROP_NET_NETWORK_NAME) tid:15
wpantund[2453]: 	↳ 8F03444F5400
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_NETWORK_NAME) tid:15
wpantund[2453]: [->NCP] CMD_PROP_VALUE_SET(PROP_NET_MASTER_KEY) tid:1
wpantund[2453]: 	↳ 81034602D74D3BBBCB9A8C2FB9EA6F09E60FCF
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_MASTER_KEY) tid:1
wpantund[2453]: [->NCP] CMD_PROP_VALUE_SET(PROP_NET_KEY_SEQUENCE_COUNTER) tid:2
wpantund[2453]: 	↳ 82034701000000
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_KEY_SEQUENCE_COUNTER) tid:0
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_STREAM_DEBUG) tid:0
wpantund[2453]: [NCP->] CMD_PROP_VALUE_IS(PROP_STREAM_DEBUG) tid:0
wpantund[2453]: NCP => Framing error 6: [03 03 03 47 01 00 00 00 EE DC]
wpantund[2453]: SpinelNCPTask.cpp:85: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[2453]: SpinelNCPTaskForm.cpp:315: Requirement Failed ((ret) == 0)
wpantund[2453]: Form failed: 7
wpantund[2453]: State change: "associating" -> "offline"
wpantund[2453]: get_property: key: "NCP:State"
wpantund[2453]: get_property: key: "Daemon:Enabled"
wpantund[2453]: NCP is no longer busy, host sleep is permitted. (Was busy for 5026ms)
```

Can you explain us why it fails to form the network?

Cheers,

abtink, 2017-06-13:
Seems to be a uart framing error here.
```
NCP => Framing error 6: [03 03 03 47 01 00 00 00 EE DC]
```

biboc, 2017-06-14:
Yes I saw it but this happen every time.
Any idea why on this specific command it fails?

biboc, 2017-06-16:
UART implementation was wrong

closed 2017-06-16 by biboc