openthread/wpantund #195: security: leave network as user? Join network as user?

DuaneEllis-TI, 2017-06-22:
Not sure of the intent, but as a "user" (not root) what should wpantund allow via wpanctl?(aka dbus)
yes, this open the "big-old-can-of-worms".

aeliot, 2017-06-22:
I may be mistaken, but isn't user permissions generally handled by the system via systemd configuration files? (At least for linux running systemd)  What permissions the user should have is a system question that is going to be different depending on the applications.

DuaneEllis-TI, 2017-06-22:
Currently as a user (not root) I can:  "wpanctl  leave" a network.

I do not think systemd provides these protections,  I think one of 3 places needs to do this:

Place (A) - in wpanctl - but that does not make security sense, it could be spoofed

Place (B) - in the dbus library, via some dbus configuration file. Most likely this is the dbus library on the wpantund side. I would also describe this as "passively via wpantund" because it s done by the library or maybe some type of kernel internals?

Place (C) is actively within wpantund, in that wpantund needs to decide based upon either by (1) effective user id, and/or  (2) based on the type of action.  ie: Anyone can do "status"  but only root can change things.

This is not my area of expertise.

aeliot, 2017-06-22:
It also depends on the application.  For example, in an embedded linux product you would want your application to run in non-root space, you dont want to have to give your application root privileges, but you also need it to be able to control all the settings in wpantund.

DuaneEllis-TI, 2017-06-23:
I think this caused by something goofy in my configuration... and thus is probably not a bug.
Thus I'll close this.

closed 2017-06-23 by DuaneEllis-TI

darconeous, 2017-06-23:
DBus should be enforcing these restrictions. Mind describing how your "goofy" setup might have caused this?