openthread/wpantund #225: Get wpantund up and running on OSS-Fuzz

darconeous, 2017-09-13:
With the fuzzing-related work on #224, #222, and #191 now coming together nicely, we should get a continuous fuzzing system in place. It just so happens that Google has [OSS-Fuzz](https://github.com/google/oss-fuzz), which at first glance looks like it would be easier to get set up and running compared to the alternatives.

darconeous, 2017-09-13:
@jwhui, [OSS-Fuzz](https://github.com/google/oss-fuzz) this might be useful for OpenThread, too.

jwhui, 2017-09-13:
@darconeous, OpenThread is already added to OSS-Fuzz :)

https://github.com/google/oss-fuzz/tree/master/projects/openthread

darconeous, 2017-09-15:
Awesome!

I've got OSS-Fuzz working well on [my branch](https://github.com/darconeous/oss-fuzz/tree/wpantund-fuzz/projects/wpantund) on my local machine. Once #224 lands I'll file a pull request against OSS-Fuzz.

darconeous, 2017-09-15:
Actual OSS-Fuzz pull request: https://github.com/google/oss-fuzz/pull/843

darconeous, 2017-09-15:
https://github.com/google/oss-fuzz/pull/843 has been merged. Will close once I see it running.

darconeous, 2017-09-19:
It is now up and running.

closed 2017-09-19 by darconeous