openthread/wpantund #252: DBus Introspectable

robbym, 2017-11-26:
Does wpantund's DBus interface not implement introspectable?
I'm using a library that relies on it.

I'm new to DBus and could just be using my tools wrong, but I thought I'd ask just in case.

robbym, 2017-11-29:
Ok, so it does not implement Introspectable.
The dbus documentation also does not accurately represent the current implementation.
Some things are still under `com.nestlabs.WPANTunnelDriver` when the docs list them under `org.wpantund`.

robbym, 2017-12-01:
I have added basic introspection support here: https://github.com/robbym/wpantund
However, I feel like its more of a hack. If this were done right, I'd somehow integrate it into the existing DBus classes, with some sort of validation support based on the type description.

I was thinking of a way to specify the types using variadic templates, and have it generate introspection and validation methods, but I'm not sure the compiler version we are using is capable.

jwhui, 2019-04-23:
Closing stale issue.

closed 2019-04-23 by jwhui

darconeous, 2019-09-09:
Introspection support would be pretty cool.