openthread/wpantund #277: Invalid Neighbor during Join caused by unicast address in ncp

DuaneEllis-TI, 2018-02-22:
This is the WPANTUND side of these two issues:

https://github.com/openthread/openthread/pull/2579

https://github.com/openthread/openthread/issues/2571

In the end, the NCP is now rejecting addresses being sent to the NCP from wpantund
causing the WPANTUND to send a RESET command to the NCP...

From openthread pull request 2579 -

>> Feb 22 18:57:06 beaglebone wpantund[1344]: Adding on-mesh prefix "fd11:22::/64" to NCP
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: SpinelNCPTaskSendCommand.cpp:322: Requirement Failed ((mRetVal) == 0)
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: SendCommand task encountered an error: 24 (0x00000018)
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: Error NCPInvalidArgument (24) while performing "adding unicast address" on NCP - Resetting NCP.
>> Feb 22 18:57:06 beaglebone wpantund[1344]: Node type change: "unknown" -> "leader"
>> Feb 22 18:57:06 beaglebone wpantund[1344]: SpinelNCPTaskSendCommand.cpp:322: Requirement Failed ((mRetVal) == 0)
>> Feb 22 18:57:06 beaglebone wpantund[1344]: SendCommand task encountered an error: 24 (0x00000018)
>> Feb 22 18:57:06 beaglebone wpantund[1344]: Error NCPInvalidArgument (24) while performing "adding unicast address" on NCP - Resetting NCP.
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: Error Canceled (11) while performing "restoring on-mesh prefix" on NCP - Resetting NCP.
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: State change: "associated" -> "uninitialized"
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: Taking interface(s) down. . .
>> Feb 22 18:57:06 beaglebone wpantund[1344]: wpantund[1344]: NCPInstance: error 0

THIS is the reset

>> Feb 22 18:57:06 beaglebone wpantund[1344]: Error Canceled (11) while performing "restoring on-mesh prefix" on NCP - Resetting NCP.

>> Feb 22 18:57:06 beaglebone wpantund[1344]: State change: "associated" -> "uninitialized"

DuaneEllis-TI, 2018-02-22:
Adding: @srickardti @pkanek so they are aware.

abtink, 2018-02-27:
Submitted PR: https://github.com/openthread/wpantund/pull/279

abtink, 2018-03-02:
https://github.com/openthread/wpantund/pull/279 is merged and should address this.

abtink, 2018-05-07:
I think we can close this with the change from https://github.com/openthread/wpantund/pull/279

abtink, 2018-06-19:
Closing this issue.

closed 2018-06-19 by abtink