openthread/wpantund #281: Problems with SPI support

jciupis, 2018-03-09:
Hi,

I have been working on SPI support for nRF52840 platform in OpenThread to establish SPI connection between NCP run on a nRF52840 and OpenThread Border Router run on a Raspberry Pi 3. You can find my work here if necessary: [](https://github.com/jciupis/openthread/tree/pr/add-spi-slave-support-for-nrf52840)

I have built latest OpenThread libraries as follows:

`root@debian:/home/jeci/repo/openthread# make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 TMF_PROXY=1 SPI=1`

making sure that --with-ncp-bus=spi is chosen in Makefile-nrf52840. The version of wpantund on the border router:

```
pi@raspberrypi:~$ wpantund -v
wpantund 0.08.00d (/7fea6d7; Mar  2 2018 02:23:36)
```

With the SPI interface on the RPi3 up and running, the communication seems to be working - when I run the following command:

`pi@raspberrypi:~$ echo -ne "\x01\x02\x03" > /dev/spidev0.0`

I can see the 1, 2, 3 in the SPI callback on NCP. I have also tested transfers between two nRF52840 DKs using the SPI code I developed and it also worked fine, so as far as I know the NCP behaves correctly. However, when I try to run wpantund on SPI, I get the following logs:

```
pi@raspberrypi:~$ sudo wpantund
wpantund[882]: Starting wpantund 0.08.00d (Mar  2 2018 02:23:36) . . .
wpantund[882]:  BUILD_VERSION = 7fea6d7
wpantund[882]: Configuration file "/etc/wpantund.conf" read.
wpantund[882]: set-config-param: "Daemon:SyslogMask" set succeded
wpantund[882]: Stat (autolog): Tx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[882]: Stat (autolog): Tx history is empty
wpantund[882]: Stat (autolog):
wpantund[882]: Stat (autolog): Rx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[882]: Stat (autolog): Rx history is empty
wpantund[882]: Stat (autolog):
wpantund[882]: Stat (autolog): NCP state history is empty.
wpantund[882]: Stat (autolog):
wpantund[882]: Stat (autolog):
wpantund[882]: tcgetattr() failed. "Inappropriate ioctl for device" (25)
wpantund[882]: tcsetattr() failed. "Inappropriate ioctl for device" (25)
wpantund[882]: Opening tun interface socket with name "wpan0"
wpantund[882]: Getting DBus connection
wpantund[882]: Registering DBus connection
wpantund[882]: Requesting DBus connection name com.nestlabs.WPANTunnelDriver
wpantund[882]: Ready. Using DBUS bus ":1.45"
wpantund[882]: Running as root without dropping privileges!
wpantund[882]: Initializing NCP
wpantund[882]: Removing all NCP originated addresses
wpantund[882]: NCP is now BUSY.
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: [->NCP] CMD_PROP_VALUE_GET(PROP_PROTOCOL_VERSION) tid:1
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: SpinelNCPInstance-Protothreads.cpp:412: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: Resetting and trying again... (retry 1)
wpantund[882]: [->NCP] CMD_RESET tid:2
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[882]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
```

Current content of /etc/wpantund.conf is left to default apart from the socket part:

```
# Example wpantund configuration file
#

# The desired name of the network interface (not supported on all platforms)
# Default value is `wpan0`.
#
#Config:TUN:InterfaceName wpan0

# Path to serial port used to communicate with the NCP.
# Has special meaning when prefixed with `system:` or `serial:`.
# If the path is an IPv4 address/port, it will use a TCP socket.
#
#Config:NCP:SocketPath "/dev/ttyACM0"
#Config:NCP:SocketPath "/dev/tty.usbmodem1234"
#Config:NCP:SocketPath "127.0.0.1:4901"
Config:NCP:SocketPath "/dev/spidev0.0"
#Config:NCP:SocketPath "serial:/dev/ttyO1,raw,b115200,crtscts=1"
```

I had to change the path to the SPI socket from the original:

`#Config:NCP:SocketPath "system:/usr/local/sbin/spi-server -p - -s /dev/spidev2.0"`

because spi-server doesn't exist, as shown in the logs below.

```
root@raspberrypi:/home/pi# wpantund
wpantund[942]: Starting wpantund 0.08.00d (Mar  2 2018 02:23:36) . . .
wpantund[942]:  BUILD_VERSION = 7fea6d7
wpantund[942]: Configuration file "/etc/wpantund.conf" read.
wpantund[942]: set-config-param: "Daemon:SyslogMask" set succeded
wpantund[942]: Stat (autolog): Tx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[942]: Stat (autolog): Tx history is empty
wpantund[942]: Stat (autolog):
wpantund[942]: Stat (autolog): Rx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[942]: Stat (autolog): Rx history is empty
wpantund[942]: Stat (autolog):
wpantund[942]: Stat (autolog): NCP state history is empty.
wpantund[942]: Stat (autolog):
wpantund[942]: Stat (autolog):
wpantund[942]: Opening tun interface socket with name "wpan0"
wpantund[944]: Forked!
wpantund[944]: About to exec "/usr/local/sbin/spi-server -p - -s /dev/spidev0.0"
wpantund[942]: Getting DBus connection
/bin/bash: /usr/local/sbin/spi-server: No such file or directory
wpantund[942]: Registering DBus connection
wpantund[942]: Requesting DBus connection name com.nestlabs.WPANTunnelDriver
wpantund[942]: Ready. Using DBUS bus ":1.56"
wpantund[942]: Running as root without dropping privileges!
wpantund[942]: Initializing NCP
wpantund[942]: Removing all NCP originated addresses
wpantund[942]: NCP is now BUSY.
wpantund[942]: [-NCP-]: Socket error on read: Input/output error 5
wpantund[942]: NCPInstance: errno 5 "Input/output error"
wpantund[942]: Cleaning up. (gRet = 9)
wpantund[942]: Stopped.
```

Where do you think the problem is? How can I proceed further to make it work?

abtink, 2018-03-09:
`"/dev/spidev0.0"` cannot be directly used as `Config:NCP:SocketPath`. `wpantund` expects the socket to provide HDLC encoded spinel frames. 

[`spi-hdlc-adapter`](https://github.com/openthread/openthread/tree/master/tools/spi-hdlc-adapter) is the the tool that can be used on the host (SPI master) side to send/receive Spinel frames over SPI to/form NCP and perform HDLC encoding. 

`wpantund` allows the `Config:NCP:SocketPath` to be set to a system executable. We can then use `spi-hdlc-adapter`. Here is an example of this in `wpantund.conf` (assuming `spi-hdlc-adapter` is installed at path `/usr/sbin/`)
```
Config:NCP:SocketPath "system:/usr/sbin/spi-hdlc-adapter --stdio -i <path to the I̅N̅T̅ pin if any> -r <path to R̅E̅S̅ pin if any>  --spi-speed=<spi-speed default is 1MHz> <dev path to spi>
```
`spi-hdlc-adapter` has more config options, please see [`README`](https://github.com/openthread/openthread/tree/master/tools/spi-hdlc-adapter).

jciupis, 2018-03-14:
Thank you very much, it works like a charm now. I would suggest slightly modifying the documentation though - if the `spi-hdlc-adapter` is necessary, the example you have provided in your answer seems to be much better than the one in the current version:

`#Config:NCP:SocketPath "system:/usr/local/sbin/spi-server -p - -s /dev/spidev2.0"`

abtink, 2018-05-07:
Submitted https://github.com/openthread/wpantund/pull/297 to add spi-hdlc-adapter example in the provided `wpantund.conf` sample