openthread/wpantund #286: Native USB CDC communication issues after NCP reset

rlubos, 2018-03-15:
Hello! I've been testing OpenThread Border Router on RPi and NCP with native USB CDC for communication (nRF52840), and have experienced the following issue.

When issuing a "leave" command, the NCP is being reset, which is completely fine. But as I am using native USB CDC for communication, reseting the device makes it reenumarate in the system and in result the communication with the device is broken, which causes wpantund crash.

```
root@raspberrypi:/home/pi# wpanctl -I wpan0 leave                         
Leaving current WPAN. . .                                                 
leave failed with error 11. Canceled                                      
                                                                          
DIAGNOSIS: This action was aborted due to a change in the NCP's state.    
This can occur if the interface is disabled while you were trying to join,
or if AutoDeepSleep kicked in for some reason.                            
```
Syslog entries:
```
Mar 14 20:59:24 raspberrypi wpantund[22416]: wpantund[22416]: State change: "associated" -> "offline"
Mar 14 20:59:24 raspberrypi wpantund[22416]: wpantund[22416]: Resetting interface(s). . .
Mar 14 20:59:24 raspberrypi wpantund[22416]: State change: "associated" -> "offline"
Mar 14 20:59:24 raspberrypi wpantund[22416]: wpantund[22416]: Node type change: "leader" -> "unknown"
Mar 14 20:59:24 raspberrypi wpantund[22416]: Resetting interface(s). . .
Mar 14 20:59:24 raspberrypi wpantund[22416]: Node type change: "leader" -> "unknown"
Mar 14 20:59:25 raspberrypi wpantund[22416]: [-NCP-]: Socket error on read: Broken pipe 32
Mar 14 20:59:25 raspberrypi wpantund[22416]: wpantund[22416]: [-NCP-]: Socket error on read: Broken pipe 32
Mar 14 20:59:25 raspberrypi wpantund[22416]: wpantund[22416]: NCPInstance: errno 11 "Resource temporarily unavailable"
Mar 14 20:59:25 raspberrypi wpantund[22416]: wpantund[22416]: Cleaning up. (gRet = 9)
Mar 14 20:59:25 raspberrypi wpantund[22416]: NCPInstance: errno 11 "Resource temporarily unavailable"
Mar 14 20:59:25 raspberrypi wpantund[22416]: Cleaning up. (gRet = 9)
Mar 14 20:59:25 raspberrypi wpantund[22416]: wpantund[22416]: Stopped.
Mar 14 20:59:25 raspberrypi wpantund[22416]: Stopped.
Mar 14 20:59:25 raspberrypi systemd[1]: wpantund.service: Main process exited, code=exited, status=9/n/a
Mar 14 20:59:25 raspberrypi systemd[1]: wpantund.service: Unit entered failed state.
Mar 14 20:59:25 raspberrypi systemd[1]: wpantund.service: Failed with result 'exit-code'.
Mar 14 20:59:30 raspberrypi systemd[1]: wpantund.service: Service hold-off time over, scheduling restart.
Mar 14 20:59:30 raspberrypi systemd[1]: Stopped wpantund.
```

After wpantund is up again, everything works fine and I can configure Thread network correctly. It is an issue though when I try to use Border Router's web app to form a network, because a failure during "leave" causes entire form operation to fail.

I'm not very familiar with wpantund itself so I have a question to the experts - would it be possible to reopen the socket after device is reset and how difficult would it be to implement it? Or should I rather seek for solution at higher layer (simply ignore the error code)?

jwhui, 2018-03-15:
@rlubos, what a coincidence!  @darconeous is in the process of addressing this issue.  In fact, he just submitted a proposal (https://github.com/openthread/openthread/pull/2621), please review and comment.

rlubos, 2018-03-15:
Wow, I'm in shock 😄 I'm glad to hear that it's already been adressed, will give it a look :+1:,

darconeous, 2018-03-15:
https://github.com/openthread/openthread/pull/2621 has been merged!

closed 2018-03-15 by darconeous