openthread/wpantund #304: Update tag "full/latest-release"

zhanglongxia, 2018-05-31:
I follow the installation guide in wpantund/INSTALL.md to install wpantund. Then I run the script ./start.sh in /openthread/tests/toranj, the output shows that the wpantund doesn't support property "MAC:Whitelist:Enabled". After I reinstall the wpantund using the latest code, this issue disappear. I think we may need to update the tag "full/latest-release" or change the section 2.1 of wpantund/INSTALL.md.

Here is the log:
[log.txt](https://github.com/openthread/wpantund/files/2057414/log.txt)

Here is the git log of the wpantund:
```
zhanglongxia-macbookpro:wpantund zhanglongxia$ git log
commit 6993264fd613fbc67378bdf686ab608657cb77a9 (HEAD -> latest_0_7_1, tag: full/latest-release, tag: full/0.07.01)
Merge: 0097803 fd66805
Author: Robert Quattlebaum <rquattlebaum@nestlabs.com>
Date:   Fri Mar 31 13:50:16 2017 -0700

    build-system: Merge of changes for 0.07.01
    
            Based-on: 85d5e61b0138b362d02e244e122c8f5aaddafe09

commit fd6680530ac77813067d1d7c4c3090ffc1736568 (origin/autoconf/release/0.07/master)
Author: Robert Quattlebaum <rquattlebaum@nestlabs.com>
Date:   Fri Mar 31 13:50:15 2017 -0700

    build-system: Autotools rebuild from 0.07.01
    
    Based-on: 85d5e61b0138b362d02e244e122c8f5aaddafe09
```

Here is the git log of the openthread.
```
zhanglongxia-macbookpro:openthread zhanglongxia$ git log
commit 64fed1ed4deb2a5bbcac4624dcc9222f53583a7a (HEAD -> master, origin/master, origin/HEAD)
Author: Jonathan Hui <jonhui@nestlabs.com>
Date:   Tue May 29 16:30:43 2018 -0700

    [cli] fix router table command (#2747)

commit 45d360fcd3531b19c256b48bf1b318d9712c278e
Author: Yakun Xu <bukepo@gmail.com>
Date:   Wed May 30 06:56:42 2018 +0800

    [ncp] enable missing properties in radio only mode (#2751)
```

abtink, 2018-06-12:
Thanks for reporting. `toranj` requires wpantund master.
Pushed this PR to update the readme: https://github.com/openthread/openthread/pull/2784

abtink, 2018-09-14:
The `toranj` docs are updated. Closing this issue.

closed 2018-09-14 by abtink