openthread/wpantund #311: wpanctl error: Did not receive a reply.

osrm101, 2018-06-15:
I just started with openthread and Thread in general. I tried to set up a border router on Raspi3 with CC2652 as the NCP. I started from the git repo openthread/borderrouter, but the `./script/setup` failed to install wpantund for many times. The error was `Failed to enable unit: File /etc/systemd/system/wpantund.service already exists.` I found `/etc/systemd/system/wpantund.service` uses path as `/usr/sbin/wpantund`. Then I installed latest-release wpantund from this repo, and manually edited `wpantund.service` pathes to `/usr/local/sbin/wpantund`. I am able to start wpantund service. 

Now I come to a new problem. 
`sudo /usr/local/bin/wpanctl`
`error: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
/usr/local/bin/wpanctl: error: wpantund is either not running, locked up, or incompatible with this version of wpanctl.`

I appreciate your knowledge and experience on this issue. Thanks.

abtink, 2018-06-19:
@osrm101 I am not sure what's exactly happening here. I can give some pointers:
- Can you please check if `wpantund` is running ok? (`ps -A | grep wpantund`) - or checking wpantund syslogs to see if all is ok.
- Can you try starting `wpanctl` with the specific interface like `/usr/local/bin/wpanctl -I wpan1` (replace `wpan1` with interface name from `wpantund` call).

yz101-osrm, 2018-06-19:
@abtink Thanks for your note. 
I rolled back the raspbian-stretch-lite image from 2018-04-18 to 2018-03-13, reinstalled the openthread/borderrouter, this problem doesn't exist anymore.

abtink, 2018-06-19:
Thanks @osrm101. Good to hear. I will close this issue.

closed 2018-06-19 by abtink

jwhui, 2018-06-22:
I just tested a fresh install on Raspbian Stretch Lite 2018-04-18 and the border router setup on a RPi 3B Rev 1.2 and it worked for me.

With a keyboard directly attached to the RPi and Ethernet providing Internet connectivity:
```bash
$ sudo apt-get update
$ sudo apt-get install git

$ git clone https://github.com/openthread/borderrouter.git
$ cd borderrouter

$ ./script/bootstrap
$ ./script/setup
```