openthread/wpantund #314: Garbage in Thread:Leader:StableNetworkData

NickNamerr, 2018-06-21:
Thread:DeviceMode = 0x0F
Thread:Leader:Address = "fd11:1111:1122::ff:fe00:0"
Thread:Leader:LocalWeight = 0x40
Thread:Leader:NetworkData = [08040B0200010314004020010DB88569B2B5070211400504500B33000316008020010DB88569B2B50000000000000001070202800316008020010DB88569B2B50000000000000001070203800316008020010DB88569B2B50000000000000001070204800316008020010DB88569B2B50000000000000001070205800316008020010DB88569B2B50000000000000001070206800316008020010DB88569B2B50000000000000001070207800316008020010DB88569B2B50000000000000001070208800316008020010DB88569B2B50000000000000001070209800316008020010DB88569B2B5000000000000000107020A80]
Thread:Leader:RouterID = 0x00
Thread:Leader:StableNetworkData = [0314004020010DB88569B2B5070211400504FFFE33000316008020010DB88569B2B50000000000000001070202800316008020010DB88569B2B50000000000000001070203800316008020010DB88569B2B50000000000000001070204800316008020010DB88569B2B50000000000000001070205800316008020010DB88569B2B50000000000000001070206800316008020010DB88569B2B50000000000000001070207800316008020010DB88569B2B50000000000000001070208800316008020010DB88569B2B50000000000000001070209800316008020010DB88569B2B5000000000000000107020A80]
Thread:Leader:Weight = 0x40

Kindly advice how to clean it.

NickNamerr, 2018-06-21:
Here are steps to repeat this bug:
1) scan
2) setprop Network:Key --data 00112233445566778899AABBCCDDEEFF
3) join 1 -T 2 "OpenThreadDemo"
4) config-gateway -d 2001:db8:8568:b2b5::
5) leave
6) repeat 1-3 
7) config-gateway -v 0 2001:db8:8568:b2b5::

abtink, 2018-06-22:
Hi @NickNamerr ,
I tried your steps but cannot reproduce the issue. Here is a test-script with same steps (https://github.com/abtink/openthread/blob/test/issue26/tests/toranj/test-issue26.py#L43-L69)
```
...
$ Node2.wpanctl('join "Test" -c 12 -T r -p 0x1E47 -x 0xC0978BF9199E50A1'):
     Joining "Test" C0978BF9199E50A1 as node type "router"
     Successfully Joined!
$ Node2.wpanctl('config-gateway -v 0 2001:db8:8568:b2b5::'):
     Using prefix "2001:db8:8568:b2b5::"
     Gateway configured.
$ Node2.wpanctl('get -v Thread:Leader:NetworkData') -> '[08040B022422]'
[08040B022422]

```
Can you please provide more info?  What's the leader node? Can you check network data on the leader?

NickNamerr, 2018-06-25:
Hello, I need more time to do it.

I faced it on RHEL7.5 based platform. Actually, all weird behavior I observed only there. On Ubuntu based PC with exactly same radio-board(without reprogrammed FW) could not reproduce issue aswell.

jwhui, 2018-06-25:
@NickNamerr, can you provide more detail about your setup?  For example, what is the topology, platforms, IEEE 802.15.4 SoCs, etc.?

jwhui, 2019-04-23:
Closing stale issue.

closed 2019-04-23 by jwhui