openthread/wpantund #332: Use wpanctl

GizMo94, 2018-07-19:
Hi,

I don't really know how to explain my question so I'm going to explain what I want to do.

At the moment in order to create a network I execute wpanctl : 

> form "myNetwork"

What I would like to do it's a program (.c) that execute those command, like ./form myNetwork

It will be the same for all the other command provide by the CLI. So do you know how it's possible ? 

I'm not sure to be clear .

Thank you

abtink, 2018-07-19:
Hi @GizMo94 ,
wpantund provides a set of DBus API/signals  to control NCP/wpantund from any process/application . You can use them from your program. `wpanctl` tool itself uses the DBus APIs. I would recommend looking at the implementation of different `wpanctl` commands to see how they map and use the DBus APIs.

GizMo94, 2018-07-20:
Hi @abtink ,

Thank you for the quickly answer, I'm going to check that issue.

GizMo94, 2018-07-26:
Hi, 

I'm coming back, I think I have find the DBUS API/signal --> https://github.com/openthread/wpantund/tree/2c7f9932988996a61ea513298a2baa66b11e66a3/src/ipc-dbus

and I have check how it was mapped --> https://github.com/openthread/wpantund/blob/master/src/wpanctl/wpanctl.c

I'm not really sure I'm looking the right files.

Can you please if you can just explain how I can do a really easy .c program to just form a network from Ubuntu shell without using wpanctl.

Thanks a lot

abtink, 2018-07-26:
You can checkout the `wpanctl` form implementation itself https://github.com/openthread/wpantund/blob/master/src/wpanctl/tool-cmd-form.c#L147

In short, 
- Create a dbus `connection`
- Create a dbus `message` with the command (method call) on the interface you want
- Append the corresponding parameters to dbus message (for the method call)
- Send message and wait for reply
- Parse reply to know status/outcome of command.

closed 2019-04-23 by jwhui