openthread/wpantund #342: On-mesh prefix(es) not persisted

mateusz-, 2018-08-15:
Hello,

On-mesh prefix(es) do not persist on reboot.

```sh
$ sudo wpanctl getprop Thread:OnMeshPrefixes
Thread:OnMeshPrefixes = []
$ sudo wpanctl add-prefix --stable --on-mesh --default-route --slaac --preferred --priority 0 fd11:22::
Successfully added prefix "fd11:22::" len:64 stable:1 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med]
$ sudo wpanctl getprop Thread:OnMeshPrefixes
Thread:OnMeshPrefixes = [
	"fd11:22::              prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med]"
]
sudo reboot
# SSH back in...
$ sudo wpanctl getprop Thread:OnMeshPrefixes
Thread:OnMeshPrefixes = []
```

If I reset the `wpanctl`, things work okay.
```sh
$ sudo wpanctl getprop Thread:OnMeshPrefixes
Thread:OnMeshPrefixes = []
$ sudo wpanctl add-prefix --stable --on-mesh --default-route --slaac --preferred --priority 0 fd11:22::
Successfully added prefix "fd11:22::" len:64 stable:1 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med]
$ sudo wpanctl getprop Thread:OnMeshPrefixes
Thread:OnMeshPrefixes = [
	"fd11:22::              prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med]"
]
$ sudo wpanctl reset
Resetting NCP. . .
$ sudo wpanctl getprop Thread:OnMeshPrefixes
Thread:OnMeshPrefixes = [
	"fd11:22::              prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med]"
]
```

I'd be happy to create a pull request, but I am not entirely sure where to start. Is it a matter of persisting the data, like in this [pull request](https://github.com/openthread/wpantund/pull/313/commits/ffa790efb5d2b7ff6b7c283e1c46433997d285d6)?

This issue also seems to be open in the border router repo, openthread/borderrouter#104.

mateusz-, 2018-08-15:
Should probably give some details on my setup.

```bash
$ sudo wpanctl status
wpan0 => [
	"NCP:State" => "associated"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/20170716-00726-g8c8a953d; KW41Z; Jul 10 2018 14:32:07"
	"Daemon:Version" => "0.08.00d (/df0822f; Jul 10 2018 16:44:39)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [E470050000000000]
	"NCP:Channel" => 15
	"Network:NodeType" => "leader"
	"Network:Name" => "OpenThreadDemo"
	"Network:XPANID" => 0x1111111122222222
	"Network:PANID" => 0x1234
	"IPv6:LinkLocalAddress" => "fe80::20e7:d90c:990c:580f"
	"IPv6:MeshLocalAddress" => "fd11:1111:1122:0:9664:126a:6a1c:ed1"
	"IPv6:MeshLocalPrefix" => "fd11:1111:1122::/64"
	"com.nestlabs.internal:Network:AllowingJoin" => false
]
$ uname -a
Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux
$ cat /etc/wpantund.conf | grep "^[^#]"
Config:NCP:SocketPath "/dev/ttyACM0"

```

abtink, 2018-08-15:
Unciast IPv6 addresses, multiast addresses, on-mesh prefixes and off-mesh routes are already saved  in `wpantund` and if NCP is reset they are re-applied on NCP. Some of the user-set properties 
are also saved in a setting map to persist over NCP reset (like the change in  https://github.com/openthread/wpantund/pull/313/commits/ffa790efb5d2b7ff6b7c283e1c46433997d285d6). All the info is saved by the `wpantund` process (in RAM).

But when the device is rebooted, all process are started again including `wpantund` process. This starts a fresh version of driver. 

Typically an application/network layer module is controlling the Thread network through wpantund. This module decides on addresses/prefixes/routes/etc and sets them on `wpantund`. After a reboot it makes sense for same module to re-apply the addresses/prefixes/settings (maybe from its initialization). 

An example of how this is handled by OpenWeave is the Weave Address and Routing Module (WARM) moudle: https://github.com/openweave/openweave-core/blob/master/src/warm/Warm.h

mateusz-, 2018-08-15:
That makes sense, thanks @abtink.

closed 2018-08-15 by mateusz-