openthread/wpantund #344: Routing issue or rloc16 is not updated

NickNamerr, 2018-08-31:
Faced with the issue than "wpanctl get" shows everything is set properly, but accidentally stopped routing in and out of Thread Network:

Daemon:Version = "0.08.00d (0.07.01-249-g2635e66; Aug 14 2018 15:03:11)"

**Thread:OffMeshRoutes = [
        "2001:db8:8569:b2b7::/64    origin:user     stable:yes preference:medium  rloc:0x0000 next_hop_is_host:yes"
        "2001:db8:8569:b2b7::/64    origin:ncp      stable:yes preference:medium  rloc:0x8452 next_hop_is_host:yes"
]

Thread:OnMeshPrefixes = [
        "2001:db8:8568:b2b7::   prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x0000"
        "2001:db8:8568:b2b7::   prefix_len:64   origin:ncp      stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x8452"
]
Thread:Parent = "EE07FF4047EA8921, RLOC16:8400, Age:154, AveRssi:-60, LastRssi:-60, LQIn:3, LQOut:0"
Thread:PendingDataset = []
Thread:RLOC16 = 0x8452**

after applied "wpanctl reset"

_Thread:OffMeshRoutes = [
        "2001:db8:8569:b2b7::/64    origin:user     stable:yes preference:medium  rloc:0x0000 next_hop_is_host:yes"
        "2001:db8:8569:b2b7::/64    origin:ncp      stable:yes preference:medium  rloc:0x8455 next_hop_is_host:yes"
]
Thread:OnMeshPrefixes = [
        "2001:db8:8568:b2b7::   prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x0000"
        "2001:db8:8568:b2b7::   prefix_len:64   origin:ncp      stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x8455"
]
Thread:Parent = "EE07FF4047EA8921, RLOC16:8400, Age:15, AveRssi:-59, LastRssi:-60, LQIn:3, LQOut:3"
Thread:PendingDataset = []
Thread:RLOC16 = 0x8455_

And traffic begins to route in and out automatically.
Could you help me to resolve it?

librasungirl, 2018-08-31:
@NickNamerr , thanks for reporting this.
It looks to me the link between the parent and the child became asymmetrical as
`Thread:Parent = "EE07FF4047EA8921, RLOC16:8400, Age:154, AveRssi:-60, LastRssi:-60, LQIn:3, LQOut:0" `

For my understanding, the child should only set the its parent's LQOut as 0 when it removes its parent from its neighbor in OT's implementation.

Would you please check whether there is something wrong in the keep-alive process from logs or pcap if you kept?  I guess the device become detached when the issue happens.

However even it is detached once, it should attempt to reattach afterwards. I think the logs or pcap could help to reveal what's going on during the issue happens. Thanks.

NickNamerr, 2018-08-31:
@librasungirl is there any instructions for "keep-alive process from logs or pcap" how to do it properly?

librasungirl, 2018-08-31:
If the child is non-sleepy device, the keep-alive message should be `MLE Child Update Request message/MLE Child Update Response` message.  
If there is something wrong for this process, you may find consecutive `MLE Child Update Request` from child but no `MLE Child Update Response` from the pcap, or `Send Child Update Request to parent` but no `Receive Child Update Response from parent` from logs if you have enabled "INFO" or "DEBUG" logging level when you build the NCP firmware.

Please refer to Thread 1.1 Spec `Section 4.7.5 Timing Out Children` for more technical details.

closed 2019-04-23 by jwhui