openthread/wpantund #345: getting error undefined reference to `otNcpInit'

Kabariya, 2018-09-06:
I am trying to configure Open-thread to create a simulator. I am following this official page to do this but the First thing I have noticed I got an error when I run cmd as they have given 

$ ./bootstrap
$ ./configure --enable-ncp --enable-cli --disable-docs --disable-tests --with-examples=posix

here I have looked into the code and found out that we have to run cmd like this 

$ ./bootstrap
$ ./configure --enable-ncp-app --enable-cli-app --disable-docs --disable-tests --with-examples=posix 

to run it whithout error

after that, Now when I am trying to run "make" cmd it is giving these error. And I using Ubuntu 16.04 on Virtual Box on top of windows 10.

    Making all in ncp
    CCLD     ot-ncp-ftd
    ot_ncp_ftd-main.o: In function `main':
    /home/parth/openthread-master/examples/apps/ncp/main.c:84: undefined reference to `otNcpInit'
    ../../../examples/platforms/posix/libopenthread-posix.a(libopenthread_posix_a-uart-posix.o): In function `platformUartProcess':
    /home/parth/openthread-master/examples/platforms/posix/uart-posix.c:265: undefined reference to `otPlatUartReceived'
    /home/parth/openthread-master/examples/platforms/posix/uart-posix.c:283: undefined reference to `otPlatUartSendDone'
     collect2: error: ld returned 1 exit status
     Makefile:1235: recipe for target 'ot-ncp-ftd' failed
     make[4]: *** [ot-ncp-ftd] Error 1
     Makefile:485: recipe for target 'all-recursive' failed 
     make[3]: *** [all-recursive] Error 1
     Makefile:491: recipe for target 'all-recursive' failed
     make[2]: *** [all-recursive] Error 1
     Makefile:652: recipe for target 'all-recursive' failed
     make[1]: *** [all-recursive] Error 1
     Makefile:578: recipe for target 'all' failed
     make: *** [all] Error 2

jwhui, 2018-09-06:
Unfortunately, that wiki page is out-of-date. Try following the [Simulating a Thread network using OpenThread codelab](https://codelabs.developers.google.com/codelabs/openthread-simulation/).

Kabariya, 2018-09-06:
thanks :) Now I closing it :)

closed 2018-09-06 by Kabariya