openthread/wpantund #346: how to pass messages between two simulated openthread on windows 10 with c language.

Kabariya, 2018-09-10:
I am new to opwnthread, however, I figured it out how to simulate thread I am able to pass messages between them with cmd but I want to pass it now with C language program.

jwhui, 2018-09-10:
You can try to follow the [OpenThread on Windows readme](https://github.com/openthread/openthread/tree/master/examples/drivers/windows).

Note, however, that OpenThread's Windows integration has not been actively maintained for some time. While Windows builds are still a part of OpenThread's continuous integration, the Windows continuous integration only test basic compilation. We welcome anyone who is willing to help actively maintain OpenThread's Windows support!

Kabariya, 2018-09-11:
okay... so what about Linux how can I do this thing into Linux?

jwhui, 2018-09-11:
wpantund brings up a standard linux network interface. To bring up wpantund with OpenThread in simulation, check out the [OpenThread Simulation codelab](https://codelabs.developers.google.com/codelabs/openthread-simulation). Once the network interfaces are up, you can use standard linux sockets programming to send messages between different devices.

Kabariya, 2018-09-11:
Thanks again... l was thinking something like this yesterday and then I had to go so couldn't check it out :)

jwhui, 2018-09-11:
[Zephyr](https://www.zephyrproject.org/) and [RIOT](https://riot-os.org/) both actively support OpenThread.

Silicon vendors that support OpenThread have various examples. For example, take a look at [Nordic's nRF5 SDK for Thread examples](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.threadsdk.v0.8.0%2Fexamples_thread.html).

Kabariya, 2018-09-11:
okay thanks for me out 👍

Kabariya, 2018-09-11:
we can create as many as ot-cli-ftd we want but how many ot-ncp-ftd can we create? 
I used to create wpantund with cmd 

$ sudo /usr/local/sbin/wpantund \
    -o Config:NCP:SocketPath "system:output/x86_64-unknown-linux-gnu/bin/ot-ncp-ftd 2" \
    -o Daemon:SyslogMask " -info" \
    -o Config:TUN:InterfaceName utun6

and now I want one more ncp so ran 

$ sudo /usr/local/sbin/wpantund \
    -o Config:NCP:SocketPath "system:output/x86_64-unknown-linux-gnu/bin/ot-ncp-ftd 3" \
    -o Daemon:SyslogMask " -info" \
    -o Config:TUN:InterfaceName utun7

but when I check wpanctl for both it gives me same 

"Daemon:Version"

	"NCP:Version" => "OPENTHREAD/0.01.00; POSIX; Sep  7 2018 14:45:38"
	"Daemon:Version" => "0.08.00d (; Sep  5 2018 15:53:37)"

	"NCP:Version" => "OPENTHREAD/0.01.00; POSIX; Sep  7 2018 14:45:38"
	"Daemon:Version" => "0.08.00d (; Sep  5 2018 15:53:37)

and one more thing it makes the second NCP to the leader not end device

I guess maybe it shows the same thing cause I am using one system an trying to create two wpantund.

is it possible to run two wpantund on the same system with two NCP?

closed 2019-04-23 by jwhui