openthread/wpantund #364: Border router changed RLOC16 but leader keeps hold netdata for 4 days

NickNamerr, 2018-10-16:
**Describe the bug**
Border router changed RLOC16, but leader keeps hold networkdata for 4 days

**To Reproduce**
Information to reproduce the behavior, including:
1. 2635e66
2. efr32mg13p732f512gm48
3. https://github.com/openthread/wpantund/blob/master/INSTALL.md#wpantund-installation-guide
4. Mesh

**Expected behavior**
Reclaim old networkdata within timeout or update RLOC16 in routing parameters.

**Console/log output**
Thread:Leader:NetworkData = [0314004020010DB88568B2B7070211400504789B7300030F004020010DB88569B2B70103789B00]
Thread:Leader:RouterID = 0x31
Thread:Leader:StableNetworkData = [0314004020010DB88568B2B7070211400504FFFE7300030F004020010DB88569B2B70103FFFE00]
Thread:Leader:Weight = 0x40
Thread:NeighborTable = [
	"7A0FE189985E7DC3, RLOC16:b000, LQIn:3, AveRssi:-74, LastRssi:-74, Age:0, LinkFC:1117892, MleFC:394854, IsChild:no, RxOnIdle:yes, FFD:yes, SecDataReq:no, FullNetData:yes"
]
Thread:NeighborTable:ErrorRates = [
	"7A0FE189985E7DC3, RLOC16:b000, FrameErrRate:0.00%, MsgErrorRate:0.00%, AveRssi:-74, LastRssi:-74, "
]
Thread:NetworkData = []
Thread:NetworkDataVersion = 0x6D
Thread:OffMeshRoutes = [
	"2001:db8:8569:b2b7::/64    origin:ncp      stable:yes preference:medium  rloc:0x789b next_hop_is_host:no "
]
Thread:OnMeshPrefixes = [
	"2001:db8:8568:b2b7::   prefix_len:64   origin:ncp      stable:yes flags:0x73 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:high] rloc:0x789b"
]
Thread:Parent = "EE07FF4047EA8921, RLOC16:7800, Age:4, AveRssi:-56, LastRssi:-56, LQIn:3, LQOut:0"
Thread:PendingDataset = []
Thread:RLOC16 = 0x7804

	"NCP:State" => "associated"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/20170716-00757-g3041ddf4-dirty; EFR32; Aug 14 2018 11:35:36"
	"Daemon:Version" => "0.08.00d (0.07.01-249-g2635e66; Aug 14 2018 15:03:11)"

**Additional context**
Add any other context about the problem here.

jwhui, 2018-10-16:
Can you indicate the commit id for openthread/openthread?

NickNamerr, 2018-10-17:
@jwhui 3041ddf4, I posted it above.

jwhui, 2018-10-17:
@NickNamerrl, https://github.com/openthread/openthread/commit/3041ddf4 does not exist.

NickNamerr, 2018-10-17:
Sorry, 1075e353 from 19 Jul

jwhui, 2018-10-17:
I believe this issue was addressed in https://github.com/openthread/openthread/pull/2905. Can you try using a later commit that incorporates this fix?

NickNamerr, 2018-10-17:
@jwhui I am doing it right now. I need couple days to test it and after i will update status. Thanks.

closed 2018-10-29 by NickNamerr