openthread/wpantund #373: Not able to join leader(node) from NCP

bahubalioffice67, 2018-12-04:
GitHub Issues are for bugs and feature requests. To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

The openthread-users Google Group is the recommended place for users to discuss OpenThread and interact directly with the OpenThread community. https://groups.google.com/forum/#!forum/openthread-users

bahubalioffice67, 2018-12-04:
I have two TI CC1352 launchpads. One is acting as node(cli-ftd) and other one acting as NCP. NCP is connected to linux machine(running on virtual box). Using WPANTUND I am trying to scan and join . I am able to scan node and able to see it, when I try to join it throws error as
 "wpanctl:wpan0> scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
 1 |       NO | "bahubali"         | 0xB1F2 | 21 | 54CDAAA0D3CE08CA | 8A7F19EF3762F9F6 |  -53
jwpanctl:wpan0> join 1
Joining WPAN "bahubali" as node type "end-device", channel:21, panid:0xB1F2, xpanid:0x54CDAAA0D3CE08CA [scanned network index 1]
join failed with error 10. Already
Error 10 (0x0A)"
Am I missing something?

jwhui, 2018-12-04:
@bahubalioffice67, it appears your NCP is already joined to a network. In this case, you should issue the `wpanctl leave` command before issuing `wpanctl join` again.

bahubalioffice67, 2018-12-05:
@jwhui  Thanks! Now I am able to connect

closed 2019-04-23 by jwhui