openthread/wpantund #388: Unable to add on-mesh prefix

ntavish, 2019-03-14:
**Description**

As soon as I click the "Add" button in otbr-web settings, NCP (and then wpantund) crashes.

[image]

Logs when wpantund is started:

```
wpantund[17380]: State change: "uninitialized" -> "offline"
wpantund[17380]: NCP is running "OPENTHREAD/20180926-00423-ga0db3b5b; NRF52840; Mar  1 2019 14:19:51"
wpantund[17380]: Driver is running "0.08.00d (/a1e0dfe; Nov 21 2018 21:32:05)"
wpantund[17380]: HARDWARE ADDRESS IS INVALID, MULTICAST BIT IS SET!
wpantund[17380]: Network is not joinable
wpantund[17380]: State change: "offline" -> "offline:commissioned"
wpantund[17380]: Finished initializing NCP
wpantund[17380]: AutoResume is enabled. Trying to resume.
wpantund[17380]: NCP is commissioned. Resuming...
wpantund[17380]: State change: "offline:commissioned" -> "associating"
otbr-agent[166]: NCP DBus name changed.
otbr-agent[166]: NCP property NCP:State changed.
otbr-agent[166]: state associating
otbr-agent[166]: Handle Thread change
otbr-agent[166]: Stop publishing service
otbr-agent[166]: Mdns service not available!
otbr-agent[166]: NCP property IPv6:LinkLocalAddress changed.
wpantund[17380]: State change: "associating" -> "associated"
otbr-agent[166]: NCP property NCP:State changed.
otbr-agent[166]: state associated
otbr-agent[166]: Handle Thread change
otbr-agent[166]: Start publishing service
wpantund[17380]: Node type change: "unknown" -> "end-device"
otbr-agent[166]: Avahi client state changed to 101.
otbr-agent[166]: NCP property Network:NodeType changed.
otbr-agent[166]: NCP property Network:Key changed.
otbr-agent[166]: NCP property Daemon:ReadyForHostSleep changed.
```

On attempting to add on-mesh prefix:

```
wpantund[17380]: Adding on-mesh prefix "fd11:22::/64" to NCP
wpantund[17380]: Pushing a new SLAAC address fd11:22::e8b1:cd7c:8592:7d97/64 to NCP
wpantund[17380]: Adding address "fd11:22::e8b1:cd7c:8592:7d97/64" to NCP
otbr-agent[166]: NCP property Daemon:ReadyForHostSleep changed.
wpantund[17380]: SpinelNCPTaskSendCommand.cpp:323: Requirement Failed ((mRetVal == kWPANTUNDStatus_Ok) || (mRetVal == kWPANTUNDStatus_Busy))
wpantund[17380]: SendCommand task encountered an error: 16 (0x00000010)
wpantund[17380]: SpinelNCPTaskSendCommand.cpp:381: Check Failed (error 16)
wpantund[17380]: Error PropertyNotFound (16) while performing "adding on-mesh prefix" on NCP - Resetting NCP.
otWeb[17097]: wpan service error: 8
spi-hdlc-adapter[17383]: Shutdown. (sRet = 1)
wpantund[17380]: Error Canceled (11) while performing "adding SLAAC address" on NCP - Resetting NCP.
wpantund[17380]: State change: "associated" -> "uninitialized"
wpantund[17380]: Taking interface(s) down. . .
wpantund[17380]: NCPInstance: error 0

wpantund[17380]: Resetting and trying again... (retry 1)
wpantund[17387]: About to exec "/usr/bin/spi-hdlc-adapter --stdio -i /sys/class/gpio/gpio25 --spi-speed 100000 /dev/spidev0.1"
spi-hdlc-adapter[17388]: spi-hdlc-adapter 0.07 (21:26:11 Nov 21 2018)

spi-hdlc-adapter[17388]: Slave did reset (1 resets so far)
spi-hdlc-adapter[17388]: INFO: sSlaveResetCount=1
spi-hdlc-adapter[17388]: INFO: sSpiFrameCount=8
spi-hdlc-adapter[17388]: INFO: sSpiValidFrameCount=2
spi-hdlc-adapter[17388]: INFO: sSpiDuplexFrameCount=0
spi-hdlc-adapter[17388]: INFO: sSpiUnresponsiveFrameCount=6
spi-hdlc-adapter[17388]: INFO: sSpiGarbageFrameCount=0
spi-hdlc-adapter[17388]: INFO: sHdlcTxFrameCount=0
spi-hdlc-adapter[17388]: INFO: sHdlcTxFrameByteCount=0
spi-hdlc-adapter[17388]: INFO: sHdlcRxFrameCount=1
spi-hdlc-adapter[17388]: INFO: sHdlcRxFrameByteCount=2
spi-hdlc-adapter[17388]: INFO: sHdlcRxBadCrcCount=0
wpantund[17380]: [-NCP-]: NCP was reset (RESET_SOFTWARE, 114)
wpantund[17380]: [-NCP-]: NCP was reset (RESET_SOFTWARE, 114)
wpantund[17380]: Unexpected reset during NCP initialization.
wpantund[17380]: Resetting and trying again... (retry 2)

 ---- after a few tries ----

wpantund[17380]: Unexpected reset during NCP initialization.
wpantund[17380]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[17380]: State change: "uninitialized" -> "uninitialized:fault"
spi-hdlc-adapter[17390]: INFO: sSlaveResetCount=1
spi-hdlc-adapter[17390]: INFO: sSpiFrameCount=9
spi-hdlc-adapter[17390]: INFO: sSpiValidFrameCount=4
spi-hdlc-adapter[17390]: INFO: sSpiDuplexFrameCount=1
spi-hdlc-adapter[17390]: INFO: sSpiUnresponsiveFrameCount=5
spi-hdlc-adapter[17390]: INFO: sSpiGarbageFrameCount=0
spi-hdlc-adapter[17390]: INFO: sHdlcTxFrameCount=4
spi-hdlc-adapter[17390]: INFO: sHdlcTxFrameByteCount=13
spi-hdlc-adapter[17390]: INFO: sHdlcRxFrameCount=2
spi-hdlc-adapter[17390]: INFO: sHdlcRxFrameByteCount=5
spi-hdlc-adapter[17390]: INFO: sHdlcRxBadCrcCount=0
spi-hdlc-adapter[17390]: Shutdown. (sRet = 1)
```

**To Reproduce**

1. `wpantund` is from tag `full/latest-release`
2. NCP is nRF52840, running `ot-ncp-ftd` compiled with `make -f examples/Makefile-nrf52840 NCP_SPI=1`, from master branch (a0db3b5b46b508b91f7fca3f114a56b62015cee3) 
3. `wpantund.conf` contains:
    Config:NCP:SocketPath "system:/usr/bin/spi-hdlc-adapter --stdio -i /sys/class/gpio/gpio25 --spi-speed 100000 /dev/spidev0.1"
    Config:TUN:InterfaceName wpan0 
    Daemon:SetDefaultRouteForAutoAddedPrefix true
    IPv6:SetSLAACForAutoAddedPrefix true

**Expected behavior**
`wpantund` should not crash.

ntavish, 2019-03-14:
closing as `ot-ncp-ftd` built with `make -f examples/Makefile-nrf52840 NCP_SPI=1 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1  LINK_RAW=1` works :man_facepalming:

closed 2019-03-14 by ntavish

abtink, 2019-03-14:
Yes. We need `BORDER_ROUTER` feature enabled to be able to add on-mesh prefixes.