openthread/wpantund #394: Commissioner add-joiner command fails with too many arguments on BeagleBone

theHurc, 2019-04-16:
I have built OpenThread on the BeagleBone black and it seems to work great if my devices already know the parameters to join the Thread network. However, when I go to add a device to the network via the command line with `sudo wpanctl commissioner joiner-add ...` I get an error saying there are too many arguments. I've run several commands to provide some context. If I provide 0 arguments it correctly says there are 0 arguments. If I provide 1 argument is says there are 5 arguments. And if I provide the 3 correct arguments it says there are 7 arguments.

Help please

```

phase@beaglebone:~$ sudo wpanctl status
wpan0 => [
        "NCP:State" => "associated"
        "Daemon:Enabled" => true
        "NCP:Version" => "OPENTHREAD/gefa72ce0; EFR32; Dec 14 2018 17:03:02"
        "Daemon:Version" => "0.08.00d (/d303d64; Apr 15 2019 20:24:34)"
        "Config:NCP:DriverName" => "spinel"
        "NCP:HardwareAddress" => [000D6FFFFEB88368]
        "NCP:Channel" => 21
        "Network:NodeType" => "leader"
        "Network:Name" => "p4_74E1828C5EA1"
        "Network:XPANID" => 0xEEF5C247BA2EF7DB
        "Network:PANID" => 0x49E2
        "IPv6:LinkLocalAddress" => "fe80::6c7f:f1fe:988b:2601"
        "IPv6:MeshLocalAddress" => "fd44::6a4:948:1cfb:ae2d"
        "IPv6:MeshLocalPrefix" => "fd44::/64"
        "com.nestlabs.internal:Network:AllowingJoin" => false
]
phase@beaglebone:~$ sudo wpanctl commissioner start
Commissioner started
phase@beaglebone:~$ sudo wpanctl commissioner joiner-add * 120 ABCD
commissioner joiner-add: error: Too many arguments, 7 given, expecting 3
Format:
   joiner-add <eui64> <timeout> <psk>   Add joiner, use * as <eui64> for any, <timeout> in sec (alias: `add`)
phase@beaglebone:~$ sudo wpanctl commissioner joiner-add
commissioner joiner-add: error: Too few arguments, 0 given, expecting 3
Format:
   joiner-add <eui64> <timeout> <psk>   Add joiner, use * as <eui64> for any, <timeout> in sec (alias: `add`)
phaseiv@beaglebone:~$ sudo wpanctl commissioner joiner-add *
commissioner joiner-add: error: Too many arguments, 5 given, expecting 3
Format:
   joiner-add <eui64> <timeout> <psk>   Add joiner, use * as <eui64> for any, <timeout> in sec (alias: `add`)

```

jwhui, 2019-04-16:
@theHurc, transferred to `openthread/wpantund`, as this issue seems specific to `wpanctl`.

Try adding quotes around the `*`. In particular:
```bash
$ wpanctl commissioner joiner-add "*" 120 ABCD
```

theHurc, 2019-04-16:
Oh the "*" worked!

closed 2019-04-16 by theHurc