openthread/wpantund #400: NCP out of memory. Cant find the source.

taherrera, 2019-04-23:
I am currently sending lots of information through a nRF52840 Border Router, and during this operation I receive:

`wpantund[17045]: NCP is out of memory for too long...Resetting the NCP!`

In what source code is this syslog'ed ? I cant seem to find the source file anywhere. I have also searched in the OpenThreads repository but I cant find the source of this.

Any help ?

Thanks in advance

abtink, 2019-04-23:
It seems like it's being logged by `wpantund` but I am not aware of such a log  in `wpantund` code. Doing a search I cannot find it anywhere in the source code.

Can you please provide more info... what are you running? what's your setup?

jwhui, 2019-04-23:
Seems to be from earlier versions of `wpantund`. See [release/0.07/master](https://github.com/openthread/wpantund/blob/release/0.07/master/src/ncp-spinel/SpinelNCPInstance.cpp#L986) branch, for example.

abtink, 2019-04-23:
This was added to protect against memory leaks (OT data message buffer) on NCP. I believe this was later removed. Suggest using the latest `wpantund` build.

taherrera, 2019-04-24:
Thank you @jwhui and @abtink I will update wpantund !!

I am doing pings to 4 different routers (samr21xpro running OT over RIOT) with my personal computer via a Border Router (BBB+Nordic NCP) at the "same" time.

And yes the wpantund version is:

`SOURCE_VERSION = 0.07.01-2-g6993264-dirty`

taherrera, 2019-05-02:
I installed this version of wpantund using https://github.com/openthread/ot-br-posix and following the steps in  https://openthread.io/guides/border-router/build 

After that I installed wpantund following https://github.com/openthread/wpantund/blob/master/INSTALL.md (on ubuntu).

wpantund seems to work:

`wpantund[1434]: NCP is running "OPENTHREAD/20170716-01138-g585934ea; POSIX; Apr 24 2019 18:00:29"

wpantund[1434]: Driver is running "0.07.01 (0.07.01-2-g6993264-dirty; Apr 29 2019 21:29:14)"
wpantund[1434]: Network is not joinable
wpantund[1434]: Resetting interface(s). . .
wpantund[1434]: Finished initializing NCP
wpantund[1434]: AutoResume is enabled. Trying to resume.
wpantund[1434]: NCP is commissioned. Resuming...
wpantund[1434]: State change: "offline" -> "associating"
wpantund[1434]: State change: "associating" -> "associated"`

 however, when I try to access the NCP using wpanctl I receive a timeout:

`error: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
wpanctl: error: ``wpantund`` is either not running, locked up, or incompatible with this version of ``wpanctl``.`

What steps should I take when upgrading wpantund ?

jwhui, 2019-05-31:
@taherrera, it appears you are still running an old version of `wpantund`. The version string is printing "0.07.01" but should be something like "0.08.00d".

I just did a fresh build of [ot-br-posix](https://github.com/openthread/ot-br-posix). After forming a network, this is the output:
```bash
pi@raspberrypi:~ $ sudo wpanctl status
wpan0 => [
	"NCP:State" => "associated"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/20180926-00577-g1b9c50112; NRF52840; May  5 2019 13:45:08"
	"Daemon:Version" => "0.08.00d (/8d1ce27; May 30 2019 23:37:50)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [3D537C9FE6FBDFF0]
	"NCP:Channel" => 18
	"Network:NodeType" => "leader"
	"Network:Name" => "JHUI-TEST"
	"Network:XPANID" => 0xDADEC3B0240782C0
	"Network:PANID" => 0x56FD
	"IPv6:LinkLocalAddress" => "fe80::6821:1da8:6224:7b6b"
	"IPv6:MeshLocalAddress" => "fdda:dec3:b024:0:d7aa:aa3d:6ec0:9c99"
	"IPv6:MeshLocalPrefix" => "fdda:dec3:b024::/64"
	"com.nestlabs.internal:Network:AllowingJoin" => false
]
```

jwhui, 2019-06-25:
Closing stale issue.

closed 2019-06-25 by jwhui