openthread/wpantund #416: Commissioning won't start on border router NCP

isabyr, 2019-05-30:
**Describe the bug**
I'm trying to build NCP with commissioning on border router, commissioning won't start.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id: 99937016c06674733a2b4c19abcf91cc5b4af949
2. Hardware: nrf52840/skb501
3. make -f examples/Makefile-nrf52840 NCP_SPI=1 COAP=1 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 LINK_RAW=1
4. I'm using OpenOCD with SWD pins for uploading firmware

**Expected behavior**
Instead of:
```
pi@raspberrypi:~ $ sudo wpanctl commissioner -e
Commissioner command applied.
```
I got:
```
pi@raspberrypi:~ $ sudo wpanctl commissioner -e
commissioner failed with error 16. PropertyNotFound
```
**Console/log output**
<details>
<summary>output of sudo wpanctl getprop:</summary>

```
Commissioner:ProvisioningUrl = ""
Commissioner:SessionId = 0x0000
Commissioner:State = "disabled"
Config:NCP:DriverName = "spinel"
Config:TUN:InterfaceName = "wpan0"
Daemon:AutoAssociateAfterReset = true
Daemon:AutoDeepSleep = false
Daemon:Enabled = true
Daemon:ReadyForHostSleep = true
Daemon:SetDefaultRouteForAutoAddedPrefix = true
Daemon:TerminateOnFault = false
Daemon:Version = "0.08.00d (/f19f0b8; May  2 2019 19:52:59)"
IPv6:AllAddresses = [
	"fd11:22::c091:c0a2:bce7:25cf             prefix_len:64   origin:ncp      valid:forever   preferred:forever"
	"fd11:1111:1122:0:6f64:6f6:84a8:acc7      prefix_len:64   origin:ncp      valid:forever   preferred:forever"
	"fe80::d0b4:46ea:6694:c654                prefix_len:64   origin:ncp      valid:forever   preferred:forever"
]
IPv6:LinkLocalAddress = "fe80::d0b4:46ea:6694:c654"
IPv6:MeshLocalAddress = "fd11:1111:1122:0:6f64:6f6:84a8:acc7"
IPv6:MeshLocalPrefix = "fd11:1111:1122::/64"
IPv6:MulticastAddresses = [
	"ff02::1                                  origin:ncp"
	"ff02::2                                  origin:ncp"
	"ff03::1                                  origin:ncp"
	"ff03::2                                  origin:ncp"
	"ff03::fc                                 origin:ncp"
	"ff32:40:fd11:1111:1122::1                origin:ncp"
	"ff33:40:fd11:1111:1122::1                origin:ncp"
]
IPv6:Routes = []
IPv6:SetSLAACForAutoAddedPrefix = true
Interface:Up = true
NCP:CCAFailureRate = 0x0000
NCP:CCAThreshold: <empty> (16)
NCP:Capabilities = [
	"COUNTERS (5)"
	"UNSOL_UPDATE_FILTER (12)"
	"PCAP (14)"
	"802_15_4_2450MHZ_OQPSK (24)"
	"CONFIG_FTD (32)"
	"ROLE_ROUTER (48)"
	"ROLE_SLEEPY (49)"
	"NET_THREAD_1_1 (53)"
	"MAC_RAW (513)"
	"ERROR_RATE_TRACKING (516)"
	"SLAAC (522)"
	"THREAD_COMMISSIONER (1024)"
	"THREAD_UDP_FORWARD (1026)"
	"THREAD_BORDER_ROUTER (1028)"
]
NCP:Channel = 15
NCP:ChannelMask = 134215680
NCP:Counter:AllMac = [
	"TxTotal              = 19"
	"TxUnicast            = 0"
	"TxBroadcast          = 19"
	"TxAckRequested       = 0"
	"TxAcked              = 0"
	"TxNoAckRequested     = 19"
	"TxData               = 19"
	"TxDataPoll           = 0"
	"TxBeacon             = 0"
	"TxBeaconRequest      = 0"
	"TxOther              = 0"
	"TxRetry              = 0"
	"TxErrCca             = 0"
	"TxErrAbort           = 0"
	"TxErrBusyChannel     = 0"
	"RxTotal              = 0"
	"RxUnicast            = 0"
	"RxBroadcast          = 0"
	"RxData               = 0"
	"RxDataPoll           = 0"
	"RxBeacon             = 0"
	"RxBeaconRequest      = 0"
	"RxOther              = 0"
	"RxAddressFiltered    = 0"
	"RxDestAddrFiltered   = 0"
	"RxDuplicated         = 0"
	"RxErrNoFrame         = 0"
	"RxErrUnknownNeighbor = 0"
	"RxErrInvalidSrcAddr  = 0"
	"RxErrSec             = 0"
	"RxErrFcs             = 0"
	"RxErrOther           = 0"
]
NCP:Counter:RX_IP_DROPPED = 0
NCP:Counter:RX_IP_INSEC_TOTAL = 0
NCP:Counter:RX_IP_SEC_TOTAL = 0
NCP:Counter:RX_SPINEL_ERR = 0
NCP:Counter:RX_SPINEL_TOTAL = 262
NCP:Counter:TX_IP_DROPPED = 0
NCP:Counter:TX_IP_INSEC_TOTAL = 0
NCP:Counter:TX_IP_SEC_TOTAL = 0
NCP:Counter:TX_SPINEL_TOTAL = 0
NCP:Counter:Thread:Mle = [
	"DisabledRole         = 0"
	"DetachedRole         = 1"
	"ChildRole            = 0"
	"RouterRole           = 0"
	"LeaderRole           = 1"
	"AttachAttempts       = 1"
	"PartitionIdChanges   = 1"
	"BetterPartAttaches   = 0"
	"ParentChanges        = 0"
]
NCP:ExtendedAddress = [D2B446EA6694C654]
NCP:Frequency = 2425000
NCP:HardwareAddress = [F4CE36240FE5D3B6]
NCP:MACAddress = [D2B446EA6694C654]
NCP:PreferredChannelMask = 134215680
NCP:RSSI = 1
NCP:SleepyPollInterval = 236000
NCP:State = "associated"
NCP:TXPower = 0
NCP:Version = "OPENTHREAD/20180926-00651-g99937016-dirty; NRF52840; May 30 2019 21:18:35"
Network:IsCommissioned = true
Network:Key = [00112233445566778899AABBCCDDEEFF]
Network:KeyIndex = 0
Network:Name = "OpenThreadDemo"
Network:NodeType = "leader"
Network:PANID = 0x1234
Network:PSKc = [61E1206D2C2B46E079EB775F41FC7219]
Network:PartitionId = 285413294
Network:XPANID = 0x1111111122222222
Thread:ActiveDataset = [
	"Dataset:ActiveTimestamp          =  0x0000000000000000"
	"Dataset:Channel                  =  15"
	"Dataset:NetworkName              =  "OpenThreadDemo""
	"Dataset:PanId                    =  0x1234"
	"Dataset:ExtendedPanId            =  0x1111111122222222"
	"Dataset:MasterKey                =  [00112233445566778899AABBCCDDEEFF]"
	"Dataset:MeshLocalPrefix          =  fd11:1111:1122::/64"
	"Dataset:ChannelMaskPage0         =  0x07FFF800"
	"Dataset:PSKc                     =  [61E1206D2C2B46E079EB775F41FC7219]"
	"Dataset:SecPolicy:KeyRotation    =  672"
	"Dataset:SecPolicy:Flags          =  0xFF"
]
Thread:AddressCacheTable = []
Thread:ChildTable = []
Thread:ChildTable:Addresses = []
Thread:ChildTimeout = 240
Thread:DeviceMode = 0x0F
Thread:Leader:Address = "fd11:1111:1122::ff:fe00:ec00"
Thread:Leader:LocalWeight = 0x40
Thread:Leader:NetworkData = [08040B02090203140040FD11002200000000070211400504EC003300]
Thread:Leader:RouterID = 0x3B
Thread:Leader:StableNetworkData = [03140040FD11002200000000070211400504FFFE3300]
Thread:Leader:Weight = 0x40
Thread:NeighborTable = []
Thread:NeighborTable:ErrorRates = []
Thread:NetworkData = [03100040FD110022000000000504EC003300]
Thread:NetworkDataVersion = 0xB9
Thread:OffMeshRoutes = []
Thread:OnMeshPrefixes = [
	"fd11:22::              prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x0000"
	"fd11:22::              prefix_len:64   origin:ncp      stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0xec00"
]
Thread:Parent: <empty> (15335444)
Thread:PendingDataset = []
Thread:RLOC16 = 0xEC00
Thread:RouterDowngradeThreshold = 0x17
Thread:RouterID = 0x3B
Thread:RouterTable = [
	"D2B446EA6694C654, RLOC16:ec00, RouterId:59, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:0, LinkEst:no"
]
Thread:RouterUpgradeThreshold = 0x10
Thread:StableNetworkData = [03100040FD110022000000000504EC003300]
Thread:StableNetworkDataVersion = 0x9A
com.nestlabs.internal:Network:AllowingJoin = false
com.nestlabs.internal:Network:PassthruPort = 0x1634

```
</details>

jwhui, 2019-05-30:
@isabyr, the `-e` option was removed from the commissioner command nearly a year ago (#333).

The output of `sudo wpanctl commissioner` should look like:
```bash
pi@raspberrypi:~ $ sudo wpanctl commissioner
commissioner: error: Missing command.
Syntax:
   commissioner [args] <command>
Options:
   -t/--timeout              Set timeout period [ms]
   -c/--check-psk            Checks validity/format of PSK for joiner add
Commands:
   help                                                         Print help.
   start                                                        Start Commissioner (alias: `enable`)
   stop                                                         Stop Commissioner (alias: `disable`)
   joiner-add <eui64> <timeout> <psk>                           Add joiner, use * as <eui64> for any, <timeout> in sec (alias: `add`)
   joiner-remove <eui64> [<timeout>=0]                          Remove joiner, use * as <eui64> for any, <timeout> in sec (alias: `remove`)
   announce-begin <chan_mask> <count> <period> <dest-ip>        Send Announce Begin message, <period> in millisec (alias: `announce`)
   energy-scan <chan_mask> <count> <period> <dur> <dest-ip>     Send Energy Scan Query message, <period>/<dur> in millisec (alias: `scan`)
   pan-id-query <pan-id> <chan_mask> <dest-ip>                  Send PAN ID Query message (alias: `panid`)
   mgmt-get [<tlvs>]                                            Send MGMT_COMMISSIONER_GET, <tlvs> as hex byte array
   mgmt-set <tlvs>                                              Send MGMT_COMMISSIONER_SET, <tlvs> as hex byte array
```

If you see the same output, can you try the newer version:
```
pi@raspberrypi:~ $ sudo wpanctl commissioner start
Commissioner started
```

isabyr, 2019-05-31:
@jwhui  Thanks for quick response

The output of `sudo wpanctl commissioner` `sudo wpanctl commissioner start` gives nothing:
[image]

Commissioner:State is still disabled
[image]

Are those `NCP_SPI=1 COAP=1 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 LINK_RAW=1` switches enough to enable commissioning?

jwhui, 2019-05-31:
Can you provide more information on how you setup your Raspberry Pi? Did you follow the OpenThread Border Router [Build and Configuration](https://openthread.io/guides/border-router/build) guide?

isabyr, 2019-05-31:
Yes, I have build everything(wpandtund, otbr) using official tutorials. Network is formed using via web-interface. The only warning comes when I connect NCP with SPI, I don't use interrupt pin:
[image]
Also I build firmware directly on Raspberry. I'm afraid there might be problem with hardware, I will test on other boards and report if there any changes.

abtink, 2019-05-31:
@isabyr , I think you may be be using older version of `wpanctl`. The output you have is consistent with how older `commissioner` command worked. I suggest you try `wpanctl -v` and check the version and compare it with `wpantund` version.

isabyr, 2019-06-03:
@abtink both versions are latest. I've tried on 2 other boards, they hangup on network forming step. I suspect those(skb501) are troubled boards. Is there anything else can be checked?

jwhui, 2019-06-03:
@isabyr, can you provide the output of the following commands?
- `wpanctl -v`
- `wpantund -v`
- `wpanctl status`

I do not have first-hand experience with the skb501 boards.

isabyr, 2019-06-03:
@jwhui Here is the output: 
[image]
I've built with `NCP_SPI=1 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1` switches

jwhui, 2019-06-03:
You have old versions of `wpanctl` and `wpantund` installed. The version string should start with `0.08.00d`.

I suggest starting with a fresh Raspbian image and follow the OpenThread Border Router [Build and Configuration](https://openthread.io/guides/border-router/build) guide again. The boostrap/setup scripts should install the latest versions of `wpanctl` and `wpantund`.

isabyr, 2019-06-05:
@jwhui Thank you very much, commissioner command responds now. I have made a clean installation, looks like I installed wpantund separately from ot-br-posix, so they were conflicting.
[image]

closed 2019-06-05 by isabyr