openthread/wpantund #431: How to use Restoring Function

XPAWAN, 2019-07-29:
I have started wpantund and add prefix to the network.
After wpantund restart I can see restoring logs "Restoring interface/user originated address/prefix/route entries on NCP" but the prefix is not restored. 
How to use this feature correctly?

abtink, 2019-07-29:
Hi, 
Can you please help provide more details about the commands you used to add the prefix?
Thanks

XPAWAN, 2019-08-05:
Hi,
I was using following commands
```
	wpanctl leave
	sleep 1

	wpanctl reset
	sleep 2

	wpanctl setprop Network:Key --data "00112233445566778899AABBCCDDEEFF"
	wpanctl setprop Network:PANID 0xABCD
	wpanctl setprop Network:XPANID DEAD00BEEF00CAFE
	wpanctl setprop NCP:Channel 11
	wpanctl setprop Network:Name NordicOpenThread
	wpanctl setprop Network:PSKc E00F739803E92CB42DAA7CCE1D2A394D
	wpanctl attach
```
When switching to "associated" state I was adding a new prefix with command:
```
	wpanctl config-gateway -d fd11:22::
```
When this operation is done I am restarting wpantund and expecting that after seeing log 
"Restoring interface/user originated address/prefix/route entries on NCP" this prefix is restored, but its not.

abtink, 2019-08-05:
Thanks @XPAWAN 

How are you restating `wpantund`? Note that `wpantund` does use RAM variable to save prefixes/addresses/routes, etc... and not a non-volatile memory. If you restart the `wpantund` process itself, it starts fresh.