openthread/wpantund #434: Multicast from gateway stops working after wpantund is restarted.

musalk-dk, 2019-09-09:
Consistently ping and other multicast traffic originating from the gateway does not get routed out to the thread network after the wpantund service has been restarted, stoped started.
And sometimes not after power up also.
This has been consistent problem for us using SPI mode and USB/Uart mode.
We gave up on spi mode for now since we never got multicast working (ff03::1 and ff02::1)

We have tried pulling the latest version and build it from git hop as well as 2 precompiled versions from nordic semi.
All with same issue.

We also tried to identify where mcast get's added/setup to see it we could find the fix but no luck.

bukepo, 2019-09-09:
could you please share the command you used to send the multicast traffic? Suppose your wpantund creates `wpan0`, you should use the following command to ping into Thread network:
```bash
ping6 -Iwpan0 ff03::1
```

librasungirl, 2019-09-09:
@musalk-dk , is the gateway a difference device from the wpantund device?

musalk-dk, 2019-09-10:
Thanks for you quick answers.

@bukepo Adding the parm -Iwpan0 makes it work and will solve our immediate problem with our application. 
Before I used simply ping/ping6 ff03::1 which have been working for us almost always (without -Iwpan0) our application uses ff02::1 for most of the data transfers.
I find it very weird that it works without adding -Iwpan0 after clean boot but not after wpantund service restart. 
Leaving me to believe that some things are registered correctly the first time but not at consecutive starts of the service. 
Maybe i misunderstand something but slightly confusing. I have rather good knowledge of IPv4 routings and all but i do not have a great deal of experience with IPv6 yet. 
**Perhaps you can provide a reason why ff02/ff03::1 works as general term after boot but not after wpantund restart?**

@librasungirl The setup is a simple setup USB NCP on a raspberrypi all in one.

musalk-dk, 2019-09-23:
Can no one really point me to why ping6 ff03::1 works without adding interface option after bootup but not after service restart. 

Add info: this is only true for the USB version of the NCP - running version SPI it never worked (ping without -I)

For me it seams like something is registered on boot but get messed up with a service restart of wpantund. Which would make this question an issue...

It is entirely possible that it is just my limited knowledge of IPv6 i have been browsing through some of the RFC's

We did run an application for quite a while just opening a socket and sending to ff02::1 no problem but ran into problems using a border router using SPI NCP.

So i find i quite confusing as why it works in some but not all circumstances.
And again thanks for any answers.

jwhui, 2019-09-23:
@musalk-dk, I am not sure why `ff03::1` works without specifying the interface.

In general, when sending to IPv6 multicast destination, it is necessary to specify the scope zone (see [RFC 4007 Section 5](https://tools.ietf.org/html/rfc4007#section-5) for a definition of scope zone). For IPv6 scopes where the zone is defined by physical connectivity (i.e. link/interface), that is done by specifying the interface.

closed 2019-10-04 by musalk-dk