openthread/wpantund #450: Wrong "Interface:Up" value after reset to "offline:commissioned" state

konradderda, 2019-12-06:
**Describe the bug**
After issuing the 'reset' command with the auto associating disabled "Interface:Up = true" is returned by wpandtund after it receives "PROP_VALUE_IS(NET_IF_UP) [00]".

**To Reproduce**
Information to reproduce the behavior, including:
1. Use wpantund built using the latest commit (SHA: 87c90eedce0c75cb68a1cbc34ff36223400862f1)
2. Form new network
3. Set Daemon:AutoAssociateAfterReset to false
4. Reset
5. Get status, current state should be equal to "NCP:State" => "offline:commissioned"
6. Read Interface:Up property, current output is equal to Interface:Up = true

**Expected behavior**
Resulting Interface:Up should be equal to false to reflect the NCP state.

**Console/log output**
Full debug output after the reset attached.
[auto_associate_false_reset.txt](https://github.com/openthread/wpantund/files/3931129/auto_associate_false_reset.txt)

**Additional context**
Interface:Up set to true informs that we could perform MLE discovery (scan -d from wpanctl) while in "offline:commissioned" state but it is impossible when NCP's stack is actually down.

LuDuda, 2019-12-15:
@abtink any thoughts about this issue?

abtink, 2019-12-16:
This may be a bug/inconsistency under `Daemon:AutoAssociateAfterReset`.

Setting `Daemon:AutoAssociateAfterReset` as false was only intended for testing and we generally never use it in production (we want to protect against any NCP crash/reset without wpantund user being involved).  Do you have any use-case for setting it to false?

abtink, 2019-12-16:
Did a bit of code investigation.
- "Interface:Up" seems to be mirroring the "wpan" interface state (not the NCP state).
- Setting `Daemon:AutoAssociateAfterReset` false will have other side-effects (seems to be used/controlled through primary interface callbacks as well). I would again not recommend setting this to false (is intended for testing only).
- We should be able to issue discover scan `"scan -d"` in any state. If that's your goal/use-case I think we should focus on that (e.g., we can relax the check if the ScanTask where we bring NCP interface up to include "offline:commissioned" state).

konradderda, 2019-12-18:
@abtink thank you for the investigation. What I was trying to do was to perform MLE discover while in the detached state. Basically, I was trying to use wpantund to reproduce simple command sequence used from OT CLI:
1. discover (fails)
2. ifconfig up
3. discover (succeeds)

> We should be able to issue discover scan "scan -d" in any state. 
From the NCPs perspective, the IP stack must be up and as far as I checked it is controlled by the
"Interface:Up" property. Additionally, I am not sure but it seems that setting "Interface:Up" to true starts sets both SPINEL_PROP_NET_IF_UP and SPINEL_PROP_NET_STACK_UP properties. As I searched through the code I can see no option to set SPINEL_PROP_NET_IF_UP without setting SPINEL_PROP_NET_STACK_UP property. 

> "Interface:Up" seems to be mirroring the "wpan" interface state (not the NCP state).
I am not familiar with the wpantund architecture but it seems that when we try to get "Interface:Up" property we get the value from the TunnelIPv6Interface instance which changes its state when NCP does so. I think that possible solution would be to call set_online() function of NCPBase class when SPINEL_PROP_NET_STACK_UP property is handled in SpinelNCPInstance::handle_ncp_spinel_value_is() function. 

I could provide simple PR with the fix if you think that proposed solution will fix the problem (and will not cause new issues :smile:).

abtink, 2019-12-18:
Thanks @konradderda 

> What I was trying to do was to perform MLE discover while in the detached state. Basically, I was trying to use wpantund to reproduce simple command sequence used from OT CLI:
> 
> discover (fails)
> ifconfig up
> discover (succeeds)

The "discover scan" (`scan -d`) in `wpantund`  has code which brings the interface up on NCP if we are in a state where NCP state is offline (and remembers to bring it down after discover scan is over). It basically takes care of managing NCP interface for us, so its behavior is different from OT CLI scan command. The goal for it is/was to work in any state/situation (and make it simpler for user).

https://github.com/openthread/wpantund/blob/master/src/ncp-spinel/SpinelNCPTaskScan.cpp#L144-L154

We can already issue discover scan when "NCP:State" is "offline"
```
wpanctl:wpan1> status
wpan1 => [
	"NCP:State" => "offline"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/20180926-01187-g2a9f1c76d; POSIX-toranj; Dec 18 2019 10:50:27"
	"Daemon:Version" => "0.08.00d (0.07.01-326-g6b682a5-dirty; Dec 18 2019 10:31:43)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [18B4300000000001]
]

wpanctl:wpan1> scan -d
   | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+--------------------+--------+----+------------------+------------------+------
 1 | "ak-test"          | 0x5577 | 11 | A87842D3F68D53FF | 261F3FFE715DED62 |  -20
wpanctl:wpan1> 
```
And the logs show NCP interface is being brough up and down during scan:
```
wpantund[221697]: Inbound DBus message for INTERFACE "DiscoverScanStart" from ":1.121324"
wpantund[221697]: [->NCP] (2) PROP_VALUE_SET(MAC_SCAN_MASK) [0B0C0D0E0F101112...]
wpantund[221697]: [NCP->] (2) PROP_VALUE_IS(MAC_SCAN_MASK) [0B0C0D0E0F101112...]
wpantund[221697]: [->NCP] (3) PROP_VALUE_SET(THREAD_DISCOVERY_SCAN_JOINER_FLAG) [00]
wpantund[221697]: [NCP->] (3) PROP_VALUE_IS(THREAD_DISCOVERY_SCAN_JOINER_FLAG) [00]
wpantund[221697]: [->NCP] (4) PROP_VALUE_SET(THREAD_DISCOVERY_SCAN_ENABLE_FILTERING) [00]
wpantund[221697]: [NCP->] (4) PROP_VALUE_IS(THREAD_DISCOVERY_SCAN_ENABLE_FILTERING) [00]
wpantund[221697]: [->NCP] (5) PROP_VALUE_SET(THREAD_DISCOVERY_SCAN_PANID) [FFFF]
wpantund[221697]: [NCP->] (5) PROP_VALUE_IS(THREAD_DISCOVERY_SCAN_PANID) [FFFF]
wpantund[221697]: [->NCP] (6) PROP_VALUE_SET(NET_IF_UP) [01]
wpantund[221697]: [NCP->] (6) PROP_VALUE_IS(NET_IF_UP) [01]
wpantund[221697]: [-NCP-]: Interface is up
wpantund[221697]: [->NCP] (7) PROP_VALUE_SET(MAC_SCAN_PERIOD) [C800]
...
wpantund[221697]: [NCP->] (0) PROP_VALUE_IS(MAC_SCAN_STATE) [00]
wpantund[221697]: [->NCP] (9) PROP_VALUE_SET(NET_IF_UP) [00]
wpantund[221697]: [NCP->] (9) PROP_VALUE_IS(NET_IF_UP) [00]
```

Checking the code, I see it's explicitly checking for "offline" state only `mInstance->get_ncp_state() == OFFLINE`. 

I think we should relax this check so that it applies to other "offline" states (where NCP interface is not up) like "offline:commissioned". 

I will submit a PR for this.

-----------

> I am not familiar with the wpantund architecture but it seems that when we try to get "Interface:Up" property we get the value from the TunnelIPv6Interface instance which changes its state when NCP [...]

I checked the code. This is already happening: `handle_ncp_spinel_value_is()` would change the state which would call `change_ncp_state()` which in turn calls `handle_ncp_state_change()` which does this: https://github.com/openthread/wpantund/blob/master/src/wpantund/NCPInstanceBase.cpp#L1360-L1365

With regards to tun interface (wpan) we seem to have "up" and "online/running" notions. I think there may be issues related to how they are mapped/used to exposed wpan properties. If we get the chance we should look into it but I would consider it low-priority. 

In general, I would suggest using "NCP:State" directly (to check state of NCP and whether interface is ready for use) instead of "Interface:Up" property.

abtink, 2019-12-18:
Submitted PR https://github.com/openthread/wpantund/pull/451.

konradderda, 2019-12-18:
@abtink Great! Thank you for providing the fix.

Besides of MLE discovery problem - you mentioned that "Daemon:AutoAssociateAfterReset" property is designed for the development only. Is there any way to initialize the wpantund without attaching to the network? Example use case: we would like to configure some platform specific properties before any radio communication is involved (these parameters cannot be changed on "runtime").

Steps that we would like to achieve:
1. Run wpantund with a clean NCP connected (current state: offline)
2. Form a network (current state: associated)
3. Reset the wpantund (current state: offline:commissioned)
4. Configure some properties (current state: offline:commissioned)
5. Attach to the saved network (current state: associated)

Is there any safe method to use instead of Daemon:AutoAssociateAfterReset?

abtink, 2019-12-18:
Technically you can use the `Daemon:AutoAssociateAfterReset` in the way you described but as I said we have not internally used it in production (so it's not gone through same level of testing/verification). 

Can you consider configuring the properties at step 4 while device is online? Is there a reason for device to offline during the configuration of properties at step 4?

konradderda, 2019-12-18:
> Can you consider configuring the properties at step 4 while device is online?

In the current approach - we cannot do that. We are currently testing some radio driver configuration parameters that we may provide for users in the future but currently these parameters can only be changed before network interface is enabled.  It is not the major issue for us at the moment, I just gather the possible solutions to consider.

Thank you again for providing the fix and for the clarification what `Daemon:AutoAssociateAfterReset.` was intended for. We can close this issue now or when the PR is merged.

abtink, 2019-12-19:
Just merged https://github.com/openthread/wpantund/pull/451. Closing this issue. Thanks @konradderda .

closed 2019-12-19 by abtink