openthread/wpantund #469: Upgrade and Downgrade procedure

JaneFromSilabs, 2020-06-17:
**Is your feature request related to a problem? Please describe.**
I need to use an older RCP build, so I am attempting to downgrade to an older version of the OTBR right now and having trouble doing so.  Here are the steps I am taking:
```
git checkout <commit hash> (I am using 591cfebbe1a337f84bc2549505e2fc42886ab4a6)
git submodule update --recursive --init
sudo ./script/update
```
After a reboot, I cannot talk to the RCP so I suspect I am either not doing this correctly or perhaps downgrade is not supported and I need to reflash the raspi.

**Describe the solution you'd like**
Is there documentation on how to go about this?  Additionally, is there documentation of the supported update process?  I just want to make sure I'm doing this correctly.  For updating, I would expect the same process (checkout, update submodules, update script).  Is that sequence the best practice?

abtink, 2020-06-17:
@JaneFromSilabs is your  question related to `wpantund` project? We don't have a `./script/update` in wpantund.

JaneFromSilabs, 2020-06-17:
I apologize.  I've placed this in the wrong repo by mistake.

closed 2020-06-17 by JaneFromSilabs

abtink, 2020-06-17:
No worries :)