openthread/wpantund #470: wpantund failed to start on openwrt

IHIDCHAOS, 2020-06-22:
**Describe the bug**
I compiled openwrt on the Raspberry Pi, integrated ot-br-posix and wpantund.
Now otbr-agent runs successfully, but wpantund cannot start. 
I want to know how to use wpantund on openwrt,thanks.

**To Reproduce**
Information to reproduce the behavior, including:
1. ot-br-posix is https://github.com/openthread/ot-br-posix/commit/8a7c53b9faeeef96a8a7c4fb1de364cf00acf365
wpantund is https://github.com/openthread/wpantund/commit/3f108441e23e033b936e85be5b6877dd0a1fbf1c
2. RPI3B with NRF52840DK
3. add feeds from https://github.com/hosac/openwrt-feed-openthread
4. Network topology

**Expected behavior**
A clear and concise description of what you expected to happen.

**Console/log output**
```
root@OpenWrt:~# wpantund 
wpantund[2057]: Starting wpantund 0.08.00d (Jun 18 2020 20:34:09) . . .
wpantund[2057]: 	BUILD_VERSION = be56b29707
wpantund[2057]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[2057]: Configuration file "/etc/wpantund.conf" read.
wpantund[2057]: set-config-param: "Config:NCP:SocketPath" = "serial:/dev/ttyACM0,raw,b115200,crtscts=1"
wpantund[2057]: set-config-param: "Daemon:SyslogMask" = "all -info -debug"
wpantund[2057]: ../util/tunnel.c:156: Requirement Failed (0 == ioctl(fd, TUNSETIFF, (void*)&ifr))
wpantund[2057]: ../util/tunnel.c:214: Requirement Failed (0 == ioctl(fd, TUNGETIFF, (void*)&ifr))
wpantund[2057]: TunnelIPv6Interface: Couldn't get tunnel name! errno=77, File descriptor in bad state
wpantund[2057]: Ready. Using DBUS bus ":1.1"
wpantund[2057]: Running as root without dropping privileges!
wpantund[2057]: driver_to_ncp_pump: Socket error on read: File descriptor in bad state
wpantund[2057]: NCPInstance: errno 77 "File descriptor in bad state"
wpantund[2057]: Cleaning up. (gRet = 9)
wpantund[2057]: Stopped.
```

**Additional context**
wpantund.conf use `Config:NCP:SocketPath "serial:/dev/ttyACM0,raw,b115200,crtscts=1"`

abtink, 2020-06-22:
I don't know the details, but the key line seems to be `TunnelIPv6Interface: Couldn't get tunnel name! errno=77, File descriptor in bad state`
This is where the `wpan` TUN interface is being created and system fails to get its name.
Quick things to check: Are you running as `sudo`?

bukepo, 2020-06-22:
On OpenWrt, otbr-agent is managing the tun device. So I don't think you can run wpantund and otbr-agent together. Actually, we have two solutions: wpantund + NCP + otbr-agent(use the wpantund/master branch of ot-br-posix), and otbr-agent + RCP. The latter solution is now preferred and supported on OpenWrt, in which the CLI tool is ot-ctl.

IHIDCHAOS, 2020-06-23:
> On OpenWrt, otbr-agent is managing the tun device. So I don't think you can run wpantund and otbr-agent together. Actually, we have two solutions: wpantund + NCP + otbr-agent(use the wpantund/master [...]

Yes, when I stop oybr-agent, wpantund can start normally and wpanctl can also be used. thank you!

closed 2020-06-23 by IHIDCHAOS