openthread/wpantund #473: bootstrap.sh fails to find autoconf-archive package

MatthewPennock, 2020-07-23:
**Describe the bug**
The steps in INSTALL.md cannot build the latest release of wpantund.

**To Reproduce**
See Console/log output.

**Expected behavior**
The steps in INSTALL.md should be sufficient to install wpantund.
OS: Ubuntu 20.04 LTS 64-bit

**Console/log output**
```
mpennock@vm:~$ cd source/openthread/wpantund/

mpennock@vm:~/source/openthread/wpantund$ git status
HEAD detached at full/latest-release
nothing to commit, working tree clean

mpennock@vm:~/source/openthread/wpantund$ ./bootstrap.sh 
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'm4'.
libtoolize: copying file 'm4/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
 *** error: The 'autoconf-archive' package is not installed.
Use the appropriate command for your platform to install the package:

Homebrew(OS X) ....... brew install autoconf-archive
Debian/Ubuntu ........ apt-get install autoconf-archive

mpennock@vm:~/source/openthread/wpantund$ dpkg -l | grep autoconf-archive
ii  autoconf-archive                           20190106-2.1ubuntu1               all          Autoconf Macro Archive

mpennock@vm:~/source/openthread/wpantund$ git log -n1
commit 6993264fd613fbc67378bdf686ab608657cb77a9 (HEAD, tag: full/latest-release, tag: full/0.07.01)
Merge: 0097803 fd66805
Author: Robert Quattlebaum <rquattlebaum@nestlabs.com>
Date:   Fri Mar 31 13:50:16 2017 -0700

    build-system: Merge of changes for 0.07.01
    
            Based-on: 85d5e61b0138b362d02e244e122c8f5aaddafe09
```

**Additional context**
Add any other context about the problem here.

MatthewPennock, 2020-07-23:
I can build the master branch, I believe when this pull request reaches a release the problem will be resolved: https://github.com/openthread/wpantund/pull/457

BitFis, 2020-08-03:
As @MatthewPennock points out, the issue is connected by using a newer autoconf-archive. I installed an older version as follows, this helped me fix the issue on ubuntu 20.04, hope it helps you aswell;

```bash
# clone and checkout v2017.09.28 (maybe newer versions work aswell)
git clone git://git.sv.gnu.org/autoconf-archive.git
cd autoconf-archive/
git fetch origin v2017.09.28
git checkout v2017.09.28

# configure & build and install
./bootstrap.sh
./configure --prefix=/usr # default did not work for me, check where autoconf-archive is usually installing its files
make maintainer-all # update all generated files that go into a release
make && sudo make install # build and install
```