openthread/wpantund #476: routertable eui64 does not match..

slaesh, 2020-11-25:
hey all! 
I am wondering how to read those ext-adrresses.. Trying to follow it in the code, but its kinda hard..

[image]

None of those (first values in the table) UID's are known by me and doesnt look like the UID's of my chips..

What, where and why are these values transformed?

Thanks! :)

jwhui, 2020-11-25:
The IEEE 802.15.4 Extended Address and IEEE EUI-64 are not the same. The former is what Thread devices use in IEEE 802.15.4 MAC frames during normal operation. Thread 1.1.1 Section 3.2 states the following:

> The specific value provided by the device to the MAC sublayer to represent the aExtendedAddress MUST be a random generated 64-bit value different from the universally administered IEEE EUI-64 value that has been factory assigned to the Thread interface.

slaesh, 2020-11-26:
Oh, I see! Thanks for your quick and awesome response :star_struck: 

I am wondering how to identify devices in my network. I am trying to build an application based on a borderrouter and wpantund/wpanctl.
BUT .. Imagine I am now joining a new device via the commissioner, by its EUI-64.

`wpanctl commissioner start`

and

`wpanctl commissioner joiner-add 0011223344556677 120 COOLES_PW`

After that.. HOW will I get notified that the device is actually joined and which address this device got? The device itself has no HMI or anyhting..

Kinda stuck right now in this scenario..

jwhui, 2020-11-30:
> After that.. HOW will I get notified that the device is actually joined and which address this device got? The device itself has no HMI or anyhting..

@slaesh , I suggest that your application protocol provide this functionality.

jwhui, 2020-12-17:
Closing stale issue.

closed 2020-12-17 by jwhui