openthread/wpantund #478: Disable to diagnostic mode with wpanctl(nRF52811)

Webber860, 2021-01-21:
Hi,
We added new commands and generated .hex file by "make -f examples/Makefile-nrf52811 NCP_SPI=1 FULL_LOGS=1 && arm-none-eabi-objcopy -O ihex output/nrf52811/bin/ot-ncp-mtd  output/nrf52811/bin/ot-ncp-mtd.hex" then got message blow when input mfg start.
wpanctl:wpan0> mfg start
wpantund[3087]: Inbound DBus message for INTERFACE "Mfg" from ":1.2"
wpantund[3087]: [->NCP] (15) PROP_VALUE_SET(NEST_STREAM_MFG) [737461727400]
wpantund[3087]: NCP is now BUSY.
wpantund[3087]: [NCP->] (15) PROP_VALUE_IS(LAST_STATUS) [0D]
wpantund[3087]: [-NCP-]: Last status (PROP_NOT_FOUND, 13)
wpantund[3087]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
wpantund[3087]: SendCommand task encountered an error: 16 (0x00000010)
Error 16 (0x10)
<empty>wpanctl:wpan0> wpantund[3087]: NCP is no longer busy, host sleep is permitted. (Was busy for 235ms)

thanks.
Webber

abtink, 2021-01-21:
The error reported from NCP is `PROP_NOT_FOUND, 13`. This indicates that NCP firmware does not support this property.
My guess/theory is that your NCP firmware build does not enable diag feature (which is required for the `STREAM_MFG` property to be supported by spinel layer). The diag is enabled in OT by the config `OPENTHREAD_CONFIG_DIAG_ENABLE`.

Webber860, 2021-01-22:
hi Abtink,
The problem was solved by using to build code, there is another problem that what I include file from "openthread\third_party\NordicSemiconductor\libraries\fds" and already added its path to file...third_party\NordicSemiconductor\Makefile.am, but still cannot be built as below error message during build hex file,
is there any other config file need to be added as well.
Making all in ncp
  CC       ot_ncp_mtd-main.o
  CC       ot_rcp-main.o
  CCLD     ot-ncp-mtd
  CCLD     ot-rcp
../../../examples/platforms/nrf528xx/libopenthread-nrf52811.a(libopenthread_nrf52811_a-diag.o): In function `fds_test_init':
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:669: undefined reference to `fds_register'
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:675: undefined reference to `fds_init'
../../../examples/platforms/nrf528xx/libopenthread-nrf52811.a(libopenthread_nrf52811_a-diag.o): In function `fds_test_find_and_delete':
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:695: undefined reference to `fds_record_find'
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:701: undefined reference to `fds_gc'
../../../examples/platforms/nrf528xx/libopenthread-nrf52811.a(libopenthread_nrf52811_a-diag.o): In function `fds_test_write':
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:618: undefined reference to `fds_record_write'
../../../examples/platforms/nrf528xx/libopenthread-nrf52811.a(libopenthread_nrf52811_a-diag.o): In function `fds_read':
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:638: undefined reference to `fds_record_find'
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:640: undefined reference to `fds_record_open'
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:657: undefined reference to `fds_record_close'
../../../examples/platforms/nrf528xx/libopenthread-nrf52811.a(libopenthread_nrf52811_a-diag.o): In function `fds_test_find_and_delete':
/media/public01/webber/openthread/examples/../examples/platforms/nrf528xx/src/diag.c:697: undefined reference to `fds_record_delete'
collect2: error: ld returned 1 exit status
Makefile:1668: recipe for target 'ot-rcp' failed
make[5]: *** [ot-rcp] Error 1

thanks
Webber

abtink, 2021-01-22:
This seems to be related to Nordic platform code and its makefile. I am not familiar with it.

Can provide few general advice:
- Try to follow the build command examples from OpenThread repo CI (check `script/test` and other scripts/examples).
- Always good to do a `make clean` before a new build (or even better do `git clean -dfx`).

Webber860, 2021-01-26:
Hi Abtin,
Thanks for directions you provided, I have already fixed those issue but cannot generate the hex file at the last step " arm-none-eabi-objcopy: 'output/nrf52811/bin/ot-ncp-mtd': No such file" could you help check it as below,
.......
 /bin/mkdir -p '/output//include/spinel'
 /usr/bin/install -p -m 644 /media/public01/webber/openthread/examples/../src/lib/spinel/spinel.h '/output//include/spinel'
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
Making install in cli
make[5]: Nothing to be done for 'install-data-am'.
 /bin/mkdir -p '/output/nrf52811/lib'
 /usr/bin/install -p -m 644  libopenthread-cli-mtd.a '/output/nrf52811/lib'
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-cli-mtd.a )
Making install in cli
make[5]: Nothing to be done for 'install-data-am'.
 /bin/mkdir -p '/output/nrf52811/lib'
 /usr/bin/install -p -m 644  libopenthread-cli-mtd.a '/output/nrf52811/lib'
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-cli-mtd.a )
Making install in ncp
 /bin/mkdir -p '/output/nrf52811/lib'
 /usr/bin/install -p -m 644  libopenthread-ncp-mtd.a libopenthread-rcp.a '/output/nrf52811/lib'
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-ncp-mtd.a )
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-rcp.a )
make[5]: Nothing to be done for 'install-exec-am'.
make[5]: Nothing to be done for 'install-data-am'.
Making install in examples
Making install in platforms
Making install in utils
make[6]: Nothing to be done for 'install-data-am'.
 /bin/mkdir -p '/output/nrf52811/lib'
 /usr/bin/install -p -m 644  libopenthread-platform-utils.a '/output/nrf52811/lib'
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-platform-utils.a )
Making install in k32w
make[6]: Nothing to be done for 'install-data-am'.
Making install in nrf528xx
make[6]: Nothing to be done for 'install-data-am'.
 /bin/mkdir -p '/output/nrf52811/lib'
 /usr/bin/install -p -m 644  libopenthread-nrf52811.a libopenthread-nrf52811-sdk.a libopenthread-nrf52811-transport.a '/output/nrf52811/lib'
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-nrf52811.a )
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-nrf52811-sdk.a )
 ( cd '/output/nrf52811/lib' && arm-none-eabi-ranlib libopenthread-nrf52811-transport.a )
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
Making install in apps
Making install in ncp
make[6]: Nothing to be done for 'install-data-am'.
 /bin/mkdir -p '/output/nrf52811/bin'
  /bin/bash ../../../libtool   --mode=install /usr/bin/install -p ot-ncp-mtd ot-rcp '/output/nrf52811/bin'
libtool: install: /usr/bin/install -p ot-ncp-mtd /output/nrf52811/bin/ot-ncp-mtd
libtool: install: /usr/bin/install -p ot-rcp /output/nrf52811/bin/ot-rcp
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[5]: Nothing to be done for 'install-exec-am'.
make[5]: Nothing to be done for 'install-data-am'.
Making install in tests
Making install in unit
make[5]: Nothing to be done for 'install-exec-am'.
make[5]: Nothing to be done for 'install-data-am'.
make[5]: Nothing to be done for 'install-exec-am'.
make[5]: Nothing to be done for 'install-data-am'.
Making install in tools
make[5]: Nothing to be done for 'install-exec-am'.
make[5]: Nothing to be done for 'install-data-am'.
Making install in doc
make[4]: Nothing to be done for 'install-exec-am'.
make[4]: Nothing to be done for 'install-exec-am'.
arm-none-eabi-objcopy: 'output/nrf52811/bin/ot-ncp-mtd': No such file

thanks,
Webber

abtink, 2021-01-26:
@Webber860 this seems to be related to nordic platform , makefiles and/or your toolchain.
I am not familiar with it and cannot help you. Suggest maybe reaching out to the Nordic folks.

This issue here is under `wpantund` repo and was related to diag support in `wpantund`. I think this is resolved (issue was incorrect OT_CONFIG not enabling diag mode in OT build).... If it is addressed, I suggest we close this issue.

Webber860, 2021-01-26:
Hi Abtin,
Got it, do you know where I can find the definition out for error code in "wpantund[3087]: SendCommand task encountered an error: 16 (0x00000010) Error 16 (0x10)" and I will close the issue later.

many thanks

closed 2021-01-26 by Webber860

abtink, 2021-01-26:
This is a wpantund log line when it tries to send a spinel comand to NCP.
The error means that we got an error indicating the property is not supported by the NCP.
Here is the list of `wpantund` errors: 
https://github.com/openthread/wpantund/blob/master/src/wpantund/wpan-error.h#L27