openthread/wpantund #488: Unit wpantund.service could not be found.

hansdeweerd, 2021-03-11:
i am tying to set up a thread network with a raspberry pi model 4b and a wireless starting kit(brd4001A) with a brd4161A on it as a NCP

but de wpantund is not showing up as a service. i follow this instruction: https://openthread.google.cn/certification/border-router/device-setup.

jwhui, 2021-03-11:
@hansdeweerd , the main branch in openthread/ot-br-posix implements the [RCP model](https://www.youtube.com/watch?v=EDseO8GY9Cg) and does not use `wpantund`.

If you would like to follow the Border Router Guide, try using the [wpantund/master branch](https://github.com/openthread/ot-br-posix/tree/wpantund/master) in openthread/ot-br-posix.

hansdeweerd, 2021-03-11:
thank you for your reply @jwhui 

how can I install that repository?

jwhui, 2021-03-11:
`./script/setup` should invoke [`script/_wpantund`](https://github.com/openthread/ot-br-posix/blob/wpantund/master/script/_wpantund), which installs `wpantund`.

hansdeweerd, 2021-03-12:
thanks it works!

closed 2021-03-12 by hansdeweerd