openthread/wpantund #491: ping a node in thread network with the ipv6 address.

hansdeweerd, 2021-03-31:
I what to ping a node in a thread network from my local network, but I don't know how I have to set this up. 

what I have right now:

raspberry pi model 4b with Wpantund, last commit 523bdab.
efr32mg12 with NCP: border router = 1, border router = 1, commissioner = 1, UDP forward = 1.

with steps do I have to take to make the ipv6 open thread network routable from my local ipv6 network?

jwhui, 2021-04-01:
We recently released a codelab that shows how to do this - see [Thread Border Router - Bidirectional IPv6 Connectivity and DNS-Based Service Discovery](https://openthread.io/codelabs/openthread-border-router#1). However, this requires new features that are only available in main branch of [openthread/ot-br-posix](https://github.com/openthread/ot-br-posix). Unfortunately, the main branch currently implements the RCP model and does not currently support wpantund.

Is it possible for you to migrate to using main branch of [openthread/ot-br-posix](https://github.com/openthread/ot-br-posix)?

hansdeweerd, 2021-04-01:
ok thx!

closed 2021-04-01 by hansdeweerd