openthread/wpantund #493: removing devices from a thread network.

hansdeweerd, 2021-05-12:
how can I remove a device from a thread network, and is this also possible to do this from the border router?

abtink, 2021-05-12:
In `wpantund` the `wpanctl` command `leave` (and its realted DBUS API) will cause device to leave its current network.

hansdeweerd, 2021-05-17:
yes, I understand. but I mean that you can remove a sleepy node from the network by giving the border router a command to remove it, is that possible?

abtink, 2021-05-17:
I don't think it is possible within Thread spec.

If the "Mac Filter (allow-list/deny-list)" is supported on BR device, the `AllowList/DenyList` can be used to disallow BR to talk to a particular device (which causes it to get detached).

abtink, 2021-05-19:
`wpantund` provides a bunch of properties that can be used to control `Mac::Filter`.
https://github.com/openthread/wpantund/blob/820468161c87816a3e94366ebbe2a932650416e3/src/wpantund/wpan-properties.h#L202-L208

closed 2021-06-07 by jwhui