openthread/wpantund #500: Build errors on macOS

neilalexander, 2021-06-19:
A number of build errors occur when trying to build `wpantund` on macOS 11.4, Xcode 12.5. In this case, it's on commit bf45115f41ba2b8029eda174be2b93dea73b9261.

Some of the errors are related to string handling:
```
In file included from wpantund.cpp:45:
../../src/util/string-utils.h:64:9: error: use of undeclared identifier 'strncasecmp'
        return strncaseequal(str, prefix, strlen(prefix));

wpantund.cpp:323:6: error: use of undeclared identifier 'strcasecmp'
        if (strcaseequal(key, kWPANTUNDProperty_ConfigNCPSocketBaud)) {
            ^
../../src/util/string-utils.h:33:31: note: expanded from macro 'strcaseequal'
#define strcaseequal(x, y)   (strcasecmp(x, y) == 0)
                              ^
wpantund.cpp:330:13: error: use of undeclared identifier 'strcasecmp'
        } else if (strcaseequal(key, kWPANTUNDProperty_ConfigDaemonPrivDropToUser)) {
                   ^
../../src/util/string-utils.h:33:31: note: expanded from macro 'strcaseequal'
#define strcaseequal(x, y)   (strcasecmp(x, y) == 0)
                              ^
wpantund.cpp:338:13: error: use of undeclared identifier 'strcasecmp'
        } else if (strcaseequal(key, kWPANTUNDProperty_DaemonSyslogMask)) {
                   ^
../../src/util/string-utils.h:33:31: note: expanded from macro 'strcaseequal'
#define strcaseequal(x, y)   (strcasecmp(x, y) == 0)
                              ^
wpantund.cpp:341:13: error: use of undeclared identifier 'strcasecmp'
        } else if (strcaseequal(key, kWPANTUNDProperty_ConfigDaemonChroot)) {
                   ^
../../src/util/string-utils.h:33:31: note: expanded from macro 'strcaseequal'
#define strcaseequal(x, y)   (strcasecmp(x, y) == 0)
                              ^
wpantund.cpp:348:13: error: use of undeclared identifier 'strcasecmp'
        } else if (strcaseequal(key, kWPANTUNDProperty_ConfigDaemonPIDFile)) {
                   ^
../../src/util/string-utils.h:33:31: note: expanded from macro 'strcaseequal'
#define strcaseequal(x, y)   (strcasecmp(x, y) == 0)
```
... and these above errors are largely fixed by adding the below into `src/util/string-utils.h`:
```
#ifdef __APPLE__
#include <strings.h>
#endif
```

... but a number of other errors remain:
```
wpantund.cpp:172:8: error: unknown type name 'sig_t'
static sig_t gPreviousHandlerForSIGINT;
       ^
wpantund.cpp:173:8: error: unknown type name 'sig_t'
static sig_t gPreviousHandlerForSIGTERM;
       ^
wpantund.cpp:253:69: error: use of undeclared identifier 'strsignal'
        fprintf(stderr, " *** FATAL ERROR: Caught signal %d (%s):\n", sig, strsignal(sig));
                                                                           ^
wpantund.cpp:293:68: error: use of undeclared identifier 'strsignal'
        syslog(LOG_CRIT, " *** FATAL ERROR: Caught signal %d (%s):", sig, strsignal(sig));
                                                                          ^
wpantund.cpp:603:4: error: use of undeclared identifier 'getloadavg'
                        getloadavg(loadavg, 3);
                        ^
wpantund.cpp:938:8: error: use of undeclared identifier 'chroot'
                        if (chroot(gChroot) != 0) {
```

abtink, 2021-06-21:
@neilalexander  `wpantund` on mac OS  has been experimental and never fully supported and/or tested. Even if you can build it successfully I would advice against using it on mac OS in general. I am aware of some missing features/behaviors, e.g., around netif address management (there may be more).

jwhui, 2021-09-14:
Closing stale issue.

closed 2021-09-14 by jwhui

Ja5onHoffman, 2021-11-21:
@neilalexander Wondering if you were able to solve this. Same problem.