openthread/wpantund #504: wpantund on NCP?

farhangj, 2021-08-13:
Problem Statement:
OpenThread network is meant to provide _low power_ mesh networking, however the means to connect the network to the internet (i.e. via Border router) doesn't seem to support low power use cases. I.e. where wall-powered linux router isn't available.

Assume a use-case where OpenThread is not near any local network (Wifi or Ethernet), only a cell LTE-M/NB-IoT modem is available via a UART AT cmd interface to provide internet access to all nodes of the Thread network (FTDs and SEDs).  The OT network is low power, LTE-M is low power, the only thing that is not is the OTBR, running a high power linux machine (e.g. Rasberry Pi)

Has anyone heard of or considered making a low power embedded friendly version of wpantund and whatever else is required for the operation of OTBR?

From README file of wpantund "Portability across Unix-like operating systems (currently supports Linux. BSD support should be fairly trivial to add)"

does this mean one could port this to e.g. zephyr OS and run on a Nordic nRF52 fairly easily - potentially the same chip that's working as NCP? Any pointers would be great.

abtink, 2021-08-13:
Adding some hints here which hopefully help:

- `wpantund` is a linux wpan driver which should preferably run on host which is always on and active (specially if we want to behave as active border router).
-  But it can still be used on power-aware embedded (linux based) platform where the host MCU itself may sleep (go to low-power modes) 
   - `wpantund` provides mechanism to indicate whether it is busy or not (i.e., whether host can go enter low-power/sleep mode or not).  You can check `ReadyForHostSleep` and related properties/functions/signals.
   - At Nest/Google we have used this model (e.g., "Nest Secure" can run on battery and operate with a cell connection and is running `wpantund` with host going to sleep).
- We can have a model of having a host MCU and and NCP where the host itself is running an RTOS based OS In this case we need a host side driver (equivalent to `wpantund`) to interface and control the NCP. We had internal projects/products which used this model (e.g. "Nest Connect") - however the code for this is not open-sourced (and I don't expect it to be in future). Just mentioning this that it is technically possible.
- If you want everything to run on same chip, then this is [SoC model](https://openthread.io/platforms#single-chip,-multiple-interface-soc) and then there is no need for host side driver (no need for `wpantund`) instead we need border-router related functionality (e.g. IPv6 stack and ` otbr` functionality to on the OS used by the SoC).

farhangj, 2021-08-13:
@abtink thanks for the hints.

4th bullet point is interesting.

Are there any examples that use the SoC model? Or any C drivers that could help make it happen?
Zephyr does provide networking, socket and IPV6 drivers, just not fully sure how I can put all of these together. The goal would be to have the OT Thread router of the network act also as the internet router, translating the packets received via LTE-M over UART and distributing within the OT network (to the SEDs or other routers that aren't connected to a modem)

Seems like there's no "out of the box" OTBR with low power in mind.. I'm happy to do some coding, just need some more details on the architecture.

jwhui, 2021-08-13:
You can find an example of an RTOS-based border router in the ESP-IDF:

- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/openthread.html#the-openthread-border-router

Hope that helps.

farhangj, 2021-08-18:
@jwhui Thanks, but that example is connecting to a wifi router.
All I have in my network is a LTE-M modem with AT cmd interface. e.g. can connect to end points and send/receive data. but it has no routing capabilities. This LTE-M modem is connected to a Nordic processor e.g. nRF52840 which with a suitable driver - while simultaneously an FTD of the OT network. 

I'd like to be able to provide IP connectivity through this processor for all the MTDs/SEDs in the network. Is this possible using networking drivers that Zephyr provides?
I do realize the question is a bit out of scope of wpantund...

jwhui, 2021-08-21:
I presume the cellular link is just another IP-based network interface. In that case, it is just IP routing between the cellular network and Thread network and should not be much different than routing via a Wi-Fi or Ethernet interface.

As long as Zephyr supports IP forwarding between two network interfaces, it should be possible to have a Zephyr-based devices serve as a Thread border router.

jwhui, 2021-09-14:
Closing stale issue.

closed 2021-09-14 by jwhui