openthread/wpantund #505: NCP is in offline state

bhumikaajain, 2021-09-10:
Hi,
I am using Nordic's nrf52840 as NCP in UART communication.
When I attach it to Rpi based border router, I can't get the NCP up. It is constantly in offline state. After some time it changes its state to uninitialized and then to uninitialized: fault.
Please suggest what can I do?
[image]

jwhui, 2021-09-10:
This error typically indicates that `wpantund` is not able to communicate with the NCP. I would suggest validating `wpantund` is configured with the proper device path (see `Config:NCP:SocketPath`) and that UART communication is configured properly.

bhumikaajain, 2021-09-10:
yes, I have changed the configuration path in wpantund.
It is Config:NCP:SocketPath "serial:/dev/ttyAMA0,raw,b115200,crtscts=1"
What else can I try?

jwhui, 2021-09-10:
Have you separately validated UART communication via `/dev/ttyAMA0`? Can you provide more details on the hardware setup?

bhumikaajain, 2021-09-10:
yes, I have validated the communication. 
Can you specify what further details will be helpful?

bhumikaajain, 2021-09-13:
Hi,

Below are the syslog when I run wpanctl reset and wpanctl status command:
`Sep 13 12:50:33 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTask.cpp:82: Requirement Failed (IS_EVENT_FROM_NCP(event) && ( (GetInstance(this)->mInboundHeader == mLastHeader) || spinel_callback_is_reset(event, args) ))
Sep 13 12:50:33 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:50:33 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTask.cpp:82: Requirement Failed (IS_EVENT_FROM_NCP(event) && ( (GetInstance(this)->mInboundHeader == mLastHeader) || spinel_callback_is_reset(event, args) ))
Sep 13 12:50:33 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:50:33 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:50:33 Zenatix-Testing-CBBhumika wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:16 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 13 12:51:16 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:16 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:16 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 13 12:51:16 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:16 Zenatix-Testing-CBBhumika wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:21 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 13 12:51:21 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:21 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:21 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 13 12:51:21 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:21 Zenatix-Testing-CBBhumika wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:26 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTask.cpp:82: Requirement Failed (IS_EVENT_FROM_NCP(event) && ( (GetInstance(this)->mInboundHeader == mLastHeader) || spinel_callback_is_reset(event, args) ))
Sep 13 12:51:26 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:26 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:26 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTask.cpp:82: Requirement Failed (IS_EVENT_FROM_NCP(event) && ( (GetInstance(this)->mInboundHeader == mLastHeader) || spinel_callback_is_reset(event, args) ))
Sep 13 12:51:26 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:26 Zenatix-Testing-CBBhumika wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:37 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 13 12:51:37 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:37 Zenatix-Testing-CBBhumika wpantund[1331]: wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)
Sep 13 12:51:37 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 13 12:51:37 Zenatix-Testing-CBBhumika wpantund[1331]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
Sep 13 12:51:37 Zenatix-Testing-CBBhumika wpantund[1331]: SendCommand task encountered an error: 7 (0x00000007)`

bhumikaajain, 2021-09-13:
@jwhui,
I have separately verified the UART communication, using loop_test, where  I receive the hardware address of the module on sending a command. Thus UART communication is verified.

Still I am facing NCP state Offline issue.